eclipse-threadx/netxduo #114: Restarting PPP with cellular modem fails

dsilva-vd, 2022-07-06:
Hi,

I am unable to run my ppp session (with cellular modem) when turning it on for the second time.

Here is the order of events:
1. Power on the cellular modem
2. Successfully configure and connect ppp
3. Run Azure client  -> my application
4. Stop Azure client
5. Stop ppp client
6. Power down cellular modem
7. Power up cellular modem for second time
8. Start ppp client -> FAILS - nx_ppp_create() call fails.

Platform - STM32F429
X-CUBE-CELLULAR 7.0.0
NetxDuo 6.1.7

Here is the log output.  Search for "**>>>>**" in the log below for key events:

```
===== X-Cube-Cellular version : X-CUBE-CELLULAR-7.0.0 =====

CST: New State: BOOT_STATE
TC4NAG UART config: BaudRate=115200 / HW flow ctrl=1
CST: osCDS_cellular_service_init()
CST: modem state : CA_MODEM_POWER_OFF
PPPosif:ppposif_client_start
CST: CST_notif_callback (Data Cache event=0)
CST: CST_send_message: BOOT_EVENT
CST: CST_notif_callback (Data Cache event=3)
CST: CST_send_message: BOOT_EVENT
CST: autom_event = no event 
CST: AUTOM TASK: BOOT_STATE - BOOT_EVENT
CST: CST_boot_event_mngt   ->>>>>>>>>>>>>>>>>>> Start modem power up
CST: New State: MODEM_INIT_STATE
CST: CST_send_message: MODEM_INIT_EVENT
CST: AUTOM TASK: MODEM_INIT_STATE - MODEM_INIT_EVENT
CST: CST_init_state_mngt
CST: New State: MODEM_OFF_STATE
CST: Subscribe modems events
CST: osCDS_subscribe_modem_event()
CST: CST_notif_callback (Data Cache event=7)
CST: CST_send_message: PDP_ACTIVATED_EVENT
CST: AUTOM TASK: MODEM_OFF_STATE - TARGET_STATE_CMD_EVENT
CST: CST_off_state_target_cmd_state_mngt
CST: CST_notif_callback (Data Cache event=7)
CST: CST_send_message: PDP_ACTIVATED_EVENT
CST: osCDS_power_on()
CST: CST_notif_callback (Data Cache event=0)
CST: CST_send_message: BOOT_EVENT
CST: CST_notif_callback (Data Cache event=0)
CST: CST_send_message: BOOT_EVENT
CST: modem state : CA_MODEM_STATE_POWERED_ON
CST: CST_modem_sim_init
Modem Boot OK, sim slot nb 0
CST: --> CS_sim_select sim slot nb 0
CST: osCS_sim_select()
CST: --> check SIM First 
CST: osCDS_init_modem()
CST: CST_notif_callback (Data Cache event=6)
CST: osCDS_get_device_info(IMSI)
CST: osCDS_get_device_info(ICCID)
CST: --> IMSI: 302720493042387
CST: --> MCC/MNC: 302720 (2 digit MNC only)
CST: --> ICCID: 89302720396938791246
CST: --> ICCID CC/II: 3027
CST: --> Found corresponding APN: "mtrex2.apn" in MCC/MNC table
CST: CST_notif_callback (Data Cache event=6)
CST: --> call CST_modem_define_pdn with APN found in table: "mtrex2.apn"
[... 278 lines cut ...]
CST: New State: NETWORK_STATUS_OK_STATE
CST: CST_send_message: NETWORK_STATUS_OK_EVENT
CST: AUTOM TASK: NETWORK_STATUS_OK_STATE - PPP_CLOSED_EVENT
CST: AUTOM TASK: NETWORK_STATUS_OK_STATE - NETWORK_STATUS_OK_EVENT
CST: CST_attach_modem_mngt
CST: osCDS_get_net_status()
CST: --> operator_name = "ROGERS"
CST: osCDS_get_attach_status()
CST: --> attach modem OK
CST: New State: MODEM_REGISTERED_STATE
CST: CST_send_message: MODEM_ATTACHED_EVENT
CST: AUTOM TASK: MODEM_REGISTERED_STATE - MODEM_ATTACHED_EVENT
CST: CST_modem_activate_pdn_mngt
CST: New State: MODEM_PDN_ACTIVATING_STATE
CST: osCDS_set_default_pdn()
CST: osCDS_register_pdn_event()
CST: osCDS_activate_pdn()
CST: CST_send_message: PDP_ACTIVATED_EVENT
CST: AUTOM TASK: MODEM_PDN_ACTIVATING_STATE - PDP_ACTIVATED_EVENT
PPPosif:ppposif_client_config
PPPosif:ERROR: PPP instance not created   --------------->>>>>>>>>>>>>>>>>>>>>>>>>>>   PPP Session FAILS
```

1 issues I found:
- ppposif_client_config() (file: ppposif_client.c) does not update ppposif_create_done variable after successfully making PPP connection.

TiejunMS, 2022-07-07:
@dsilva-vd , thanks for reporting the issue. I suggest posting this issue on [ST community](https://community.st.com/s/topic/0TO3W0000004ZG8WAM/azurertos) first.

dsilva-vd, 2022-07-07:
Thanks @TiejunMS  I will do that.

Can you answer this question on best practice to tear down / shut down ppp interfaces as the documentation is not clear on the differences?

To configure the link these are the steps taken in the x-cube-cellular example code:
1. Create NX_IP variable
2. Call nx_ppp_create
3. Configure Linkup and Link down callbacks
4. Call nx_ip_interface attach

What calls need to be made to tear down the ppp link?

I see there is a nx_ppp_delete and a nx_ppp_restart in the documentation.  When I have turned off my modem should I ensure nx_ppp_delete() is called or can I use nx_ppp_restart().  Currently the nx_ppp_restart() is only called when the link goes down. 

I would assume the call flow would be:
1. nx_ip_interface_detach
2. nx_ppp_delete

Thanks

TiejunMS, 2022-07-08:
When the cellular service is down, you will need to call these APIs to delete PPP cleanly.
1. `nx_ip_interface_detach` if an IP interface is already bound to PPP.
2. `nx_ppp_stop`
3. `nx_ppp_delete`

dsilva-vd, 2022-07-12:
Thanks @TiejunMS

closed 2022-07-12 by dsilva-vd