eclipse-threadx/netxduo #117: Impact of SLAAC on TCP UDP data flow

RishaHolla, 2022-07-28:
Hi,

As we understand that with SLAAC IP address of DUT changes when the prefix in the RA changes, what will be the impact of this on ongoing TCP/UDP data flow? Will the data flow be interrupted?
Secondly, can one interface have more than one valid state IP address at any time? meaning all usable for making individual connections? (1 Obtained by static/1 by dhcp/2 obtained by SLAAC) as one interface maintains multiple ipv6 address array. Or only one should be valid in the array?
Third, does not netx stack support the clearing the state/sockets/connections of the SLAAC obtained IP address when the address is not more valid?

Thanks,
Risha.

RishaHolla, 2022-08-01:
Could someone pls help with this?

TiejunMS, 2022-08-02:
When an IPv6 address becomes invalid, a callback will be invoked which can be set by `nxd_ipv6_address_change_notify`. NetX will not delete or terminate TCP/UDP sockets. User application is responsible to handle TCP/UDP sockets. All sockets would be impacted except UDP client, which depends on the remote UDP server.

One IP interface only supports one IPv4 address. It can support multiple IPv6 addresses. See more details of this macro [NX_MAX_IPV6_ADDRESSES](https://docs.microsoft.com/en-us/azure/rtos/netx-duo/chapter2#ipv6-options).

RishaHolla, 2022-08-02:
"One IP interface only supports one IPv4 address. It can support multiple IPv6 addresses. See more details of this macro NX_MAX_IPV6_ADDRESSES" as quoted by you @TiejunMS I understand that one interface can support up to max 6 ipv6 addresses as defined by this macro, but can all of them or more than one of them be valid at the same time? i.e nxd_ipv6_address_state = NX_IPV6_ADDR_STATE_VALID for more than one ip address simultaneously?

TiejunMS, 2022-08-02:
> can all of them or more than one of them be valid at the same time?

Yes, they can. See example [here](https://github.com/azure-rtos/netxduo/blob/master/samples/demo_netx_duo_multihome_tcp.c).

RishaHolla, 2022-08-02:
Ok, so basically on one IP address if there is an ongoing TCP/UDP data download/upload going on, parallelly if SLAAC obtains a new IP and makes it valid, it should not interrupt the ongoing data flow rt?

TiejunMS, 2022-08-02:
For TCP socket, it will not be impacted after the connection is established. For UDP socket, since it is not connection based, it depends on the routing table. But you can use [nxd_udp_socket_source_send ](https://docs.microsoft.com/en-us/azure/rtos/netx-duo/chapter4#nxd_udp_socket_source_send)to avoid source address change for UDP socket.

RishaHolla, 2022-08-02:
ok thanks

closed 2022-10-08 by TiejunMS