eclipse-threadx/netxduo #119: Http server basic authentication empty password

MarcoTrap, 2022-08-04:
Hello,
I try to authenticate to the web server with a valid username and an empty password and the web server accept my entry.
In this situation the code checks the username and skip the password because it reaches the end of `authorization_decoded` string.
The function is `_nx_web_http_server_basic_authenticate` and the solution is:

```
while (name_ptr[i] && (i < authorization_decoded_size))
{

    /* Is there a mismatch?  */
    if (name_ptr[i] != authorization_decoded[i])
    {

        /* Name mismatch. Continue to avoid timing attack. */
        match = NX_FALSE;
    }

    /* Move to next character.  */
    i++;
}

/* Determine if everything matches.  */
if (match && (authorization_decoded[i++] == ':') && (i < authorization_decoded_size))
{

    /* Move the authorization index past the semicolon.  */
    //i++;
```

Before was: 

```
while (name_ptr[i] && (i < authorization_decoded_size))
{

    /* Is there a mismatch?  */
    if (name_ptr[i] != authorization_decoded[i])
    {

        /* Name mismatch. Continue to avoid timing attack. */
        match = NX_FALSE;
    }

    /* Move to next character.  */
    i++;
}

/* Determine if everything matches.  */
if (match && (i < authorization_decoded_size) && (authorization_decoded[i] == ':'))
{

    /* Move the authorization index past the semicolon.  */
    i++;
```
Thanks,
Marco

bo-ms, 2022-08-05:
Hi @MarcoTrap Thank you for reporting this bug, we will fix it in next release. Much thanks.

bo-ms, 2022-11-03:
Fixed it in 6.2.0 https://github.com/azure-rtos/netxduo/releases/tag/v6.2.0_rel

closed 2022-11-03 by bo-ms