eclipse-threadx/netxduo #120: ADU on IoT Hub stuck "In Progress" or "Failed" even when firmware successfully was updated OTA

t-tbrown, 2022-08-12:
Hello,

I am using ADU with the SAME54 board, using the MPLAB sample [here](https://github.com/azure-rtos/samples/tree/PublicPreview/ADU)

I followed the steps in the documentation to set up the device and the updates are being triggered successfully from IoT Hub.

IoT Hub is failing to recognize the update went through to the device, and constantly get stuck in either "In Progress" or "Failed". However, for each of these in progress and failed deployments, as reported from IoT Hub, I can verify with a serial terminal that the board is indeed receiving, downloading, and applying the firmware update successfully.
[image]

I'm not sure how to troubleshoot this further, so any advice is welcome. Thanks!

bo-ms, 2022-08-17:
Hi @t-tbrown Thanks for reporting this issue. Device reports the old version before reporting new version to service after reboot/apply, device has already updated the new firmware and there is no real issue, but "Failed" shown up since the order of reported version is incorrect. For public preview or the project you are using, could you replace the function nx_azure_iot_adu_agent_reported_properties_send() as below and try again? 

FYI: We already fixed this issue for GA and will release it in October.

```

static UINT nx_azure_iot_adu_agent_reported_properties_send(NX_AZURE_IOT_ADU_AGENT *adu_agent_ptr, UINT wait_option)
{

NX_PACKET *packet_ptr;
NX_AZURE_IOT_JSON_WRITER json_writer;
NX_AZURE_IOT_ADU_AGENT_DEVICE_PROPERTIES *device_properties = &(adu_agent_ptr -> nx_azure_iot_adu_agent_device_properties);
NX_AZURE_IOT_ADU_AGENT_UPDATE_MANIFEST_CONTENT *manifest_content = &(adu_agent_ptr -> nx_azure_iot_adu_agent_update_manifest_content);
UINT status;
UINT response_status;
UINT request_id;
ULONG reported_property_version;
UINT result_code;
UINT i;
CHAR step_property_name[8] = "step_";
UINT step_size = sizeof("step_") - 1;
UINT step_property_name_size;
UINT update_id_length;

    /* Create json writer for client reported property.  */
    status = nx_azure_iot_hub_client_reported_properties_create(adu_agent_ptr -> nx_azure_iot_hub_client_ptr, &packet_ptr, wait_option);
    if (status)
    {
        return(status);
    }

    /* Init json writer.  */
    status = nx_azure_iot_json_writer_init(&json_writer, packet_ptr, wait_option);
    if (status)
    {
        nx_packet_release(packet_ptr);
        return(status);
    }

    /* Append begin object.  */
    if (nx_azure_iot_json_writer_append_begin_object(&json_writer))
    {
        nx_packet_release(packet_ptr);
        return(NX_NOT_SUCCESSFUL);
    }

    /* Fill the ADU agent component name.  */
    status = nx_azure_iot_hub_client_reported_properties_component_begin(adu_agent_ptr -> nx_azure_iot_hub_client_ptr,
                                                                         &json_writer, 
                                                                         (const UCHAR *)NX_AZURE_IOT_ADU_AGENT_COMPONENT_NAME,
                                                                         sizeof(NX_AZURE_IOT_ADU_AGENT_COMPONENT_NAME) - 1);
    if (status)
    {
        nx_packet_release(packet_ptr);
        return(status);
    }

    /* Fill the agent property name.  */
    if (nx_azure_iot_json_writer_append_property_name(&json_writer,
                                                      (const UCHAR *)NX_AZURE_IOT_ADU_AGENT_PROPERTY_NAME_AGENT,
                                                      sizeof(NX_AZURE_IOT_ADU_AGENT_PROPERTY_NAME_AGENT) - 1))
    {
        nx_packet_release(packet_ptr);
        return(NX_NOT_SUCCESSFUL);
    }

    /* Start to fill agent property value.   */
    if (nx_azure_iot_json_writer_append_begin_object(&json_writer))
    {
        nx_packet_release(packet_ptr);
[... 223 lines cut ...]
    }

    /* Send device info reported properties message to IoT Hub.  */
    status = nx_azure_iot_hub_client_reported_properties_send(adu_agent_ptr -> nx_azure_iot_hub_client_ptr,
                                                              packet_ptr,
                                                              &request_id, &response_status,
                                                              &reported_property_version, wait_option);
    if (status)
    {
        nx_packet_release(packet_ptr);
        return(status);
    }

    /* Check the response statue for blocking.  */
    if (wait_option)
    {
        if ((response_status < 200) || (response_status >= 300))
        {
            return(NX_NOT_SUCCESSFUL);
        }
    }

    return(NX_AZURE_IOT_SUCCESS);
}
```

bo-ms, 2022-08-29:
Closing, no further response.

closed 2022-08-29 by bo-ms