eclipse-threadx/netxduo #121: Warning from GCC about uninitialized var in snpm

josesimoes, 2022-08-25:
We've just switched to GCC 11.3-rel1 and we're getting this:

```
D:/a/1/s/build/_deps/azure_rtos_netxduo-src/addons/snmp/nxd_snmp.c: In function '_nx_snmp_utility_object_data_get':
D:/a/1/s/build/_deps/azure_rtos_netxduo-src/addons/snmp/nxd_snmp.c:15275:21: error: 'total' may be used uninitialized [-Werror=maybe-uninitialized]
15275 |          if ((total == 9) &&  (object_data -> nx_snmp_object_data_type != NX_SNMP_ANS1_OCTET_STRING))
      |              ~~~~~~~^~~~~
D:/a/1/s/build/_deps/azure_rtos_netxduo-src/addons/snmp/nxd_snmp.c:15194:9: note: 'total' declared here
15194 | UINT    total;
      |         ^~~~~
```

wenhui-xie, 2022-08-25:
Thanks for reporting this bug, we will fix it in the next release.

bo-ms, 2022-11-03:
Fixed it in 6.2.0 https://github.com/azure-rtos/netxduo/releases/tag/v6.2.0_rel

closed 2022-11-03 by bo-ms