eclipse-threadx/netxduo #123: Additional records from query

casperjuul, 2022-08-29:
Hi,

Is it only possible to readout additional records when the received packet is a response ? mdns_flags = 0x8000. I can't find anything in rfc that says query can't contain additional records so is this intentionally?

https://github.com/azure-rtos/netxduo/blob/72dcde94184b9996894c4be3b4a5cafb2b448100/addons/mdns/nxd_mdns.c#L7920-L7926

BR

bo-ms, 2022-09-05:
Hi @casperjuul Sorry for the late response. In [RFC6762, Section18.14](https://www.rfc-editor.org/rfc/rfc6762#section-18.14), query message may contain Known Answers in the Answer Section, but there is no other info that mentions query can contain additional records. In [RFC6763 Section 12](https://www.rfc-editor.org/rfc/rfc6763#section-12), server/responder should include additional records in response, it is also not for query. Did you encounter any issue for query?

bo-ms, 2022-11-03:
Closing, feel free to create new one if you have further issue.

closed 2022-11-03 by bo-ms