eclipse-threadx/netxduo #125: SLAAC with DHCPv6 interfering the DHCPv6 state

RishaHolla, 2022-09-23:
Hi,

There are 3 bits in the RA msg
M  O in the ICMPv6 flags
A in the prefix option flag

Ideally only when M bit is set the DHCPv6 server ip addressing should be triggered in stack.
1. M bit is 0 and O bit is 1 and A bit its 1 - SLAAC should be used for ip addressing and for DNS server address DHCPv6 server should be contacted
2. M bit is 0 and O bit is 0 and A bit its 1 - SLAAC should be used for ip addressing and for DNS server address probably should be obtained from RA options.
3. M bit is 1 and O bit is 1/0 and A bit is 1 - SLAAC and DHCPv6 both can be used for IP addressing and DNS sever address can be got from DHCPv6 server (O can be ignored)

But in netx stack when we run ipconfig in dhcpv6 mode, automatically DHCPv6 client is created and DHCPv6 process of obtaining ip address begins without checking for M flag in RA(scenario 1 M is 0 still DHCPv6 is started). This is because the icmpv6 is enabled in our stack only after ipconfig begins, hence no RA is processed before IP config is triggered. Now the DHCPv6 state will be NX_DHCPV6_STATE_SENDING_SOLICIT. But slaac is also triggered along with DHCPv6 and whenever there is a RA with A flag set we successfully finish slaac and in scenario 1 above since O bit is set we send out information request method asking for DNS server address and we change the dhcpv6_state to NX_DHCPV6_STATE_SENDING_INFORM_REQUEST, with this we disrupt the DHCPv6 state which was in NX_DHCPV6_STATE_SENDING_SOLICIT and hence no success/failure notification is sent to host.

This situation can be avoided i.e slaac trying to do DNS query through DHCPv6 and altering dhcpv6_state if 
1. We enabled icmpv6 for IPv6 before ipconfig from host.
2. Check for the M flag in RA and only when its set to 1 start the DHCPv6 ip addressing.

As this makes SLAAC DNS query and DHCPv6 addressing mutually exclusive events. Can we do this change?  Or is there something else we need to do?

Thanks,
Risha

bo-ms, 2022-09-27:
@RishaHolla Thanks for asking the question. I agree on your judgement for the three bits (M, O and A), but netxduo does not automatically create and start DHCPv6 Client, applications/users take responsibility for it. 

If user/application calls [nxd_icmpv6_ra_flag_callback_set()](https://github.com/azure-rtos/netxduo/blob/master/common/inc/nx_api.h#L3489-L3490) function to register an application callback, netxduo will invoke the [callback function](https://github.com/azure-rtos/netxduo/blob/master/common/src/nx_icmpv6_process_ra.c#L140-L144) to notify the application if the M, O is set or not after receiving the RA , then application can check the bits and start DHCPv6 Client. I think the logic of checking these bits or starting DHCPv6 is in user code, not in netxduo. Did you directly use our source code? Could you check the application logic?

RishaHolla, 2022-09-27:
Hi,

Unfortunately no application or demo files that uses nxd_icmpv6_ra_flag_callback_set() api is available in the repo for reference, nor could i find a reference in host scanning these bits and deciding based on them. 
Currently, in _nx_icmpv6_process_ra() api added a code like this
if((ra_ptr->nx_icmpv6_ra_flag & 0xc0) == 0x40) /* if M bit is 0 and O bit its 1 flag will be 0x40 when rest bits zeroed */
     {
        ipv6_dns_other_configuration = 1;
     }
after SLAAC completes i.e DAD is success if ipv6_dns_other_configuration flag is set due to the above lines of code, we have added call to nx_dhcpv6_request_inform_request() and make a DNS server address request with DHCPv6 server.

But again if we do not call nx_dhcpv6_request_inform_request() due to the above flag not set  and if DHCPv6 server addressing initiated (even if M flag is 0) fails then it deletes all sockets and if we have initiated a TCP data transfer based on slaac address its stops ( both slaac and DHCPv6 occupy ipv6 address 1 position in client_ip). If nx_dhcpv6_request_inform_request() gets called due to above flag set then it changes the dhcpv6_state from NX_DHCPV6_STATE_SENDING_SOLICIT  to NX_DHCPV6_STATE_SENDING_INFORM_REQUEST and even if DHCPv6 addressing fails it dosent complete and sockets are intact and TCP flow based on slaac continues. This conflict is due to DHCPv6 initiated even if M bit is 0 in RA.

bo-ms, 2022-09-28:
Hi @RishaHolla 1. DHCPv6 is addon component, netx core can not automatically start DHCPv6, 2. User/application can register the callback and check these bits in application code, then determine if start DHCPv6 or not. Is there any issue to implement this logic in application code? 3. Is it possible to share your code? then I may provide some suggestions.

RishaHolla, 2022-09-28:
Yes there is an issue to start this from application code. The issue is we get these M,O and A bits in RA and we get RA only when icmpv6 is enabled, and icmpv6 gets enabled only when dhcpv6 config is sent from host. But we want to send dhcpv6 config from host/application only when we check the M bit and find its set. So its a circle. Solution is we should enable icmpv6 before ipconfig, maybe when we find that the tcp_ip_feature bitmap has ipv6 feature set. Can we do this change?

nx_icmpv6_process_ra.c : line no 206
nx_icmpv6_perform_DAD.c : 243

[slaaccode.zip](https://github.com/azure-rtos/netxduo/files/9661453/slaaccode.zip)

bo-ms, 2022-09-28:
I am confused. Application is able to control the order of enabling icmpv6 and starting dhcpv6 config, could you show your application code where you call nxd_ipv6_enable(), nxd_icmp_enable(), nx_dhcpv6_start(), etc?

bo-ms, 2022-11-03:
Closing, feel free to reopen it.

closed 2022-11-03 by bo-ms