eclipse-threadx/netxduo #126: TLS handshake is failing for SHA384 ciphersuites

ifx-anudeep, 2022-09-29:
Hi Team,

For some feasibility testing we are using netxduo in IoT related sample and we are trying to enable SHA384(TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384 and TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384) chiper suits by adding the configs in table '_nx_crypto_ciphersuite_lookup_table_ecc' in 'nx_crypto_generic_ciphersuites.c' file.
But the TLS handshake fails after the client sends the encrypted message. From Wireshark capture, we noticed the server sent an Alert with the message 'Bad Record MAC (20)'. On debugging client code we found that api '_nx_secure_tls_record_hash_calculate()' in 'nx_secure_tls_record_hash_calculate.c' hard-codes 'output_length_in_byte' value as 'NX_SECURE_TLS_MAX_HASH_SIZE(32)' and calls crypto operations. For SHA384 hash method, it invokes '_nx_crypto_method_sha512_operation' and here api returns error 'NX_CRYPTO_INVALID_BUFFER_SIZE' as output_length_in_byte(32 Bytes) is lesser then 48 Bytes for NX_CRYPTO_HASH_SHA384.
We suspect this error is causing the TLS handshake failure. So we wanted to know if SHA384 and SHA512 cipher suits are supported in release 'v6.1.7_rel'?

Thanks
Anudeep R P

yanwucai, 2022-10-08:
SHA384 and SHA512 cipher suites are not supported in the current release.

ifx-anudeep, 2022-10-10:
Thanks for the confirmation. Since the SHA384 was defined in 'nx_secure_tls.h' we were under the assumption that netxduo supports SHA384 cipher suites. Is there any documentation where users can find details about supported cipher suites? 

Since we depend on the below SHA384 cipher suites for one of our features, it will be good to know if you plan to support it in the near term.
TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
TLS_DHE_RSA_WITH_AES_256_GCM_SHA384 

Thanks
Anudeep R P

tstapko, 2022-10-10:
> Thanks for the confirmation. Since the SHA384 was defined in 'nx_secure_tls.h' we were under the assumption that netxduo supports SHA384 cipher suites. Is there any documentation where users can find details about supported cipher suites?

NetX TLS supports X.509 certificates and digital signatures using SHA384 and SHA512. However, due to the way TLS handles the MAC (message authentication code) hashes and session key generation, it requires different logic to support ciphersuites that end in *_SHA384 and *_SHA512 (or other hashing routines). A conscious design decision was made to stick with SHA-256 only for ciphersuites to keep the memory footprint as small as possible - NetX TLS is a deeply-embedded TLS stack so it can't be expected to support all the features of a full TLS implementation. When I was on the Azure RTOS team we had a backlog item to (compile-time conditionally) add support for larger MAC hashes but it doesn't seem like that has been implemented yet. @yanwucai might have more info.

dougbenedict, 2022-10-19:
@yanwucai is there any possibility that the conditional support for larger MAC hashes is part of 6.2? Or is that just wishful thinking?

yanwucai, 2022-10-25:
Are you going to use SHA384 in TLS 1.2 or TLS 1.3 or both?

dougbenedict, 2022-10-25:
We plan to have both TLS 1.2 and 1.3 enabled.

thangdhz, 2023-07-20:
@yanwucai Do you have plan to add these cipher suites?
TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
TLS_DHE_RSA_WITH_AES_256_GCM_SHA384

fdesbiens, 2026-10-01:
I’m closing this older issue because NetX Duo has made progress on related TLS support: #377 added RSA-PSS signature verification, #399 added RSA-PSS signing for TLS 1.3 client authentication, and #417 fixed CertificateVerify when the signature hash differs from the negotiated cipher suite hash. I still see a gap for the SHA-384 cipher suites requested here: they are absent from the supplied cipher suite tables, and TLS session hash storage remains limited to 32 bytes. I’ve opened #464 to discuss whether to implement that support or clarify the documented limit.

closed 2026-10-01 by fdesbiens