eclipse-threadx/netxduo #127: TLS : Connection failure to TLS1.2 server when enabling

krshnarajd, 2022-09-30:
Hi,

We are trying to enable TLS1.3(NX_SECURE_TLS_ENABLE_TLS_1_3) in Netx secure and test the client feature.  During testing with TLS 1.2 server, it is observed that the connection is not happening with RSA public cipher (For ex: TLS-RSA-WITH-AES-256-CBC-SHA256). Server hello is proper and picking the expected cipher, but Netx-secure client is rejecting with internal error(80). 

**Setup**:
Netx TLS client(1.3 enabled) --------- TLS1.2 server

Note: This is working properly if we disable TLS1.3 in Netx or remove all the 1.3 cipher from the table.

The table used for the cipher suites is _nx_crypto_ciphersuite_lookup_table_ecc from nx_crypto_generic_ciphersuites.c

Netx stack is setting tls_session->nx_secure_tls_1_3 based on the cipher present in the table. Since the default table has the 1.3 cipher, this flag is set. Ideally this should be reset based on the server hello response from the TLS1.2 server which is not happening for the RSA cipher.  While going through the code, it is found that falling back to 1.2 is happening only if server hello contains any extensions. But server may or may not sent it. 

Server hello with no extension indicates that the server is negotiating TLS version 1.2 or prior. This looks to be a bug in the netx code as it should fallback to the tls version from the server hello if no extension is present. In the current logic, the code still thinks TLS version is 1.3 and trying to initialize handshake hash which is failing.

Regards
Krishnaraj

likidu, 2022-11-16:
@krshnarajd This is Liya, PM from Azure RTOS. Can you drop me an email liydu@microsoft.com so we can discuss this feature in more details?

yanwucai, 2022-11-16:
This bug is fixed in the 6.2.0 release.

closed 2022-11-16 by yanwucai