eclipse-threadx/netxduo #129: FTP nx_ftp_client_create return error

mapellil, 2022-10-20:
In case of sequence of calls passing the same ftp_client the last nx_ftp_client_create() return error: 
nx_ftp_client_create(&ftp_client, .....);
....
nx_ftp_client_delete(&ftp_client, .....);
....
nx_ftp_client_create(&ftp_client, .....);
This seems to happen because the nx_ftp_client_delete() do cleanup the FTP client control block (in particular the ftp_client.nx_ftp_client_id field)
Thanks, regards

wenhui-xie, 2022-10-28:
Thanks for your report. This is a bug and we will fix it in the next release.

wenhui-xie, 2023-01-09:
This bug was fixed in the commit: https://github.com/azure-rtos/netxduo/commit/2973652d801e92615caa285ad9ee13db19a26cb9.

closed 2023-01-09 by bo-ms