eclipse-threadx/netxduo #130: PTP packet ordering issue

leobbditestcom, 2022-11-02:
When the grandmaster sends a Follow_up message immediately after a Sync message to the NetxDuo PTP client, both packets are queued to the PTP UDP sockets.  The packet processing loop reads from the general message socket before the event message socket.  This causes the Follow_up message to be processed before the corresponding Sync message.  The Follow_up message is then ignored because its message ID doesn't match that last stored from a Sync message.

Reversing the processing order in nxd_ptp_client.c fixes this problem:

```
1441,1442c1441,1442
<                 if (nx_udp_socket_receive(&(client_ptr -> nx_ptp_client_event_socket),
<                                                 &packet_ptr, NX_NO_WAIT) == NX_SUCCESS)
---
>                 if (nx_udp_socket_receive(&(client_ptr -> nx_ptp_client_general_socket),
>                                             &packet_ptr, NX_NO_WAIT) == NX_SUCCESS)
1444c1444
<                     _nx_ptp_client_process_event_packet(client_ptr, packet_ptr);
---
>                     _nx_ptp_client_process_general_packet(client_ptr, packet_ptr);
1451,1452c1451,1452
<                 if (nx_udp_socket_receive(&(client_ptr -> nx_ptp_client_general_socket),
<                                             &packet_ptr, NX_NO_WAIT) == NX_SUCCESS)
---
>                 if (nx_udp_socket_receive(&(client_ptr -> nx_ptp_client_event_socket),
>                                                 &packet_ptr, NX_NO_WAIT) == NX_SUCCESS)
1454,1455c1454
<                     _nx_ptp_client_process_general_packet(client_ptr, packet_ptr);
< 
---
>                     _nx_ptp_client_process_event_packet(client_ptr, packet_ptr);
```

TiejunMS, 2022-11-03:
@leobbditestcom, thanks for reporting the issue! It is confirmed and will be fixed in the future.

TiejunMS, 2023-12-28:
This issue is fixed by https://github.com/azure-rtos/netxduo/commit/94ccd95ef0912c0f349b92df78318f973da90394#diff-a0cb2945d218e0ff4681c378290fc4dcb5a8f40a83214ab0320d9142a9b7b9ad

closed 2023-12-28 by TiejunMS