eclipse-threadx/netxduo #134: Enterprise security support over NetXSecure

cy-jayasankar, 2022-11-08:
Hi,
We are trying to add enterprise security support with NetXDuo network stack and NetXSecure TLS stack and facing following issue.

With enterprise security, TLS handshake needs to be performed with RADIUS server using EAP packets during WiFi connection. NetXSecure API user(enterprise security code) need to get the control over the send and receive functions during the TLS handshake, so that he can create EAP packets over the TLS handshake messages. But currently NetXSecure TLS APIs expects a connected socket from the user and sends the TLS handshake message using the socket pointer. So the enterprise security code does not get control over send and receive functions and cannot create EAP packets for TLS handshake messages. Also during the WiFi connection it is not possible to provide a connected socket to TLS APIs, as the sockets come into picture after the WiFi connection.

Can you please let us know if there is a mechanism to add Enterprise security support with NetXSecure?
Also if there is some implementation of Enterprise security over NetXSecure please point me to that code. 

Thanks,
Jayasankar.

EdouardMALOT, 2024-06-18:
Hi @cy-jayasankar,

I am facing a similar issue.

Did you find a way to add EAP to NetX?
Did you switch to another TCP/IP stack to handle it ?

Thanks

hwmaier, 2024-06-18:
Did you have a look at https://github.com/Infineon/nx-secure-wpa3 ?

fdesbiens, 2026-10-01:
I reviewed the current NetX Secure TLS interface. `nx_secure_tls_session_start` requires an `NX_TCP_SOCKET`, and the TLS record path sends and receives through TCP socket functions. I could not find a supported API for carrying TLS records through EAP during Wi-Fi authentication.

I'm closing this as answered because the current API does not support the requested integration.

closed 2026-10-01 by fdesbiens