eclipse-threadx/netxduo #137: How to get notified when the DHCPv6 assigned address becomes valid?

jkuwaha, 2022-11-17:
Hi,

We are trying to start an SNTP client as soon as possible when an address assigned by a DHCPv6 client becomes valid.

By using `nxd_ipv6_address_change_notify()`, our application can get notified when the DHCPv6 client sets an IP address. But our application cannot start the SNTP client because the address is still tentative.

With the `NX_ENABLE_IPV6_ADDRESS_CHANGE_NOTIFY` option, `nxd_ipv6_address_change_notify()`  notifies the address change along with its status. However, our application cannot use the notification because the DHCPv6 client uses it.

https://github.com/azure-rtos/netxduo/blob/83b08938ad3201382c4b0c7f9a71ae1cbd2d1cd3/addons/dhcp/nxd_dhcpv6_client.c#L1390-L1394

We are using NetX Duo 6.1.10. 

Are there any means to get notified when the address becomes valid?

bo-ms, 2022-11-17:
Hi @jkuwaha Thanks for asking. You may register the DHCPv6 state change callback calling [nx_dhcpv6_client_create()](https://github.com/azure-rtos/netxduo/blob/83b08938ad3201382c4b0c7f9a71ae1cbd2d1cd3/addons/dhcp/nxd_dhcpv6_client.h#L895-#L897). Then application can get the notification of state change, such as: start SNTP client once state is changed to NX_DHCPV6_STATE_BOUND_TO_ADDRESS, refer to sample demo_netxduo_dhcpv6_client.c
https://github.com/azure-rtos/netxduo/blob/83b08938ad3201382c4b0c7f9a71ae1cbd2d1cd3/samples/demo_netxduo_dhcpv6_client.c#L636-L686

jkuwaha, 2022-11-17:
Hi @bo-ms,
Thank you for your quick reply.

Your advice is very reasonable, and we have already tried the callback you mentioned, but that didn't work, unfortunately.

We observed the followings:

1. DHCPv6 sends SOLICIT to a server.
1. DHCPv6 assigns the IP address using `nxd_ipv6_address_set` after `nx_dhcpv6_state` is changed `NX_DHCPV6_STATE_BOUND_TO_ADDRESS`.
   In `nxd_ipv6_address_set`, `nxd_ipv6_address_state` is set to `NX_IPV6_ADDR_STATE_TENTATIVE`.
1. DHCPv6 calls the registered callback via the function pointer `nx_dhcpv6_state_change_callback`.
1. Our application triggers its thread to start the SNTP client.
1. SNTP client is created and set up, then `nx_sntp_client_run_unicast` is called.
1. This triggers the SNTP client's internal thread to call `nxd_udp_socket_send`.
1. `nxd_udp_socket_send` searches an interface to use by `_nxd_ipv6_interface_find`.
1. `_nxd_ipv6_interface_find` fails to find the interface, because the address assigned by DHCPv6 is still `NX_IPV6_ADDR_STATE_TENTATIVE`, and returns `NX_NO_INTERFACE_ADDRESS`.

The problem here is that the address set by DHCPv6 is not usable until it becomes valid on DAD success.

To get notified when the address becomes valid, we were considering using `nxd_ipv6_address_change_notify`, but we cannot use it because DHCPv6 also uses it.

Are there any means to get notified when the address *set by DHCPv6* becomes *valid*?

bo-ms, 2022-11-17:
Hi @jkuwaha I just double check the code, we may need to improve the state for DAD process, such as: set the state to NX_DHCPV6_STATE_BOUND_TO_ADDRESS after DAD. Performing DAD should be done in 5 seconds, could you start SNTP client after 5 seconds for testing. We will improve the logic and get back to you.

jkuwaha, 2022-11-17:
Hi @bo-ms,
I confirmed SNTP client works with a 5-sec wait!

And, thank you for considering improving the timing of a DHCPv6 callback!

Until the improvement has done, I'd like to poll the status of the address DHCPv6 assigned, but I couldn't figure out how to get the DAD status of the address.
Are there any APIs to do this, or do I need to read directly from an NX_IP instance?

bo-ms, 2022-11-17:
Hi @jkuwaha You may poll the address_count from nx_dhcpv6_get_valid_ip_address_count() API, once the return status is NX_SUCCESS and address_count is 1, you can start SNTP.

jkuwaha, 2022-11-18:
Hi @bo-ms,

I confirmed your idea works well.
Thank you very much!

bo-ms, 2022-11-21:
Hi @jkuwaha Good to know it works. I am closing it, feel free to reopen it.

closed 2022-11-21 by bo-ms