eclipse-threadx/netxduo #138: There is a difference in the processing order in NetX Duo Telnet Server

IsaoOzawa, 2022-11-22:
There is a difference in the processing order between the two functions for disconencting a telnet connection.
The following two functions for disconencting telnet connections in "nxd_telnet_server.c" do not have the same processing order.

   _nx_telnet_server_disconnect()  
        :  Call connection end callback routine                    -> Decrement "nx_telnet_server_open_connections"

   _nx_telnet_server_disconnect_process()
       :  Decrement "nx_telnet_server_open_connections" -> Call connection end callback routine

Since the values handled in the callback function are different in the two cases, the user cannot describe correct processing in the callback function.
In my understanding, it is preferable to decrement at fiest and then call the callback function in order to detect the last connection is disconnected.

wenhui-xie, 2022-11-23:
@IsaoOzawa Thanks for your report. You are right that it's better to update the "nx_telnet_server_open_connections" before calling the connection end callback function. We will fix this bug in the next release.

wenhui-xie, 2022-12-19:
Fixed in the latest update: https://github.com/azure-rtos/netxduo/commit/2973652d801e92615caa285ad9ee13db19a26cb9

closed 2022-12-19 by bo-ms