eclipse-threadx/netxduo #142: TLS 1.3: Packet Leak in Receive Path.

Anjali-Manikonda, 2023-01-18:
Hi,

We are trying to enable NetXSecure with TLS 1.3 on our platforms. As part of our testing, we found that a testcase that does TLS Client Connect -> Send -> Receive -> Disconnect is causing a packet leak and if this test is run in a loop, we eventually run out of packets.
With our debugging, we observed that during receive, we use nx_secure_tls_session_receive() api, which internally invokes _nx_secure_tls_record_chained_packet_decrypt(). This function allocates packets for both application data and Post Handshake messages, but the packets allocated for Post Handshake messages are not getting freed and causing the packet leak. In our test code, after Send -> Receive, we also invoke nx_secure_tls_session_end() and nx_secure_tls_session_delete(), but still these packets allocated for Post Handshake Messages are not freed. 

We don't see any such issue with TLS 1.2, this might be because there are no Post Handshake messages involved in TLS 1.2.
We used NetXDuo-v6.2.0 as TLS 1.3 client (NX_SECURE_TLS_ENABLE_TLS_1_3 Enabled) and a Local Python Server (Python 3.7.7).

yanwucai, 2023-01-19:
Thanks for your report, we are looking into this issue.

cy-jayasankar, 2023-03-14:
Hi..Can you please let us know if the fix for this issue is available in [v6.2.1_rel](https://github.com/azure-rtos/netxduo/releases/tag/v6.2.1_rel). If not please let us know when will it be available.

yanwucai, 2023-03-15:
It is available in [v6.2.1_rel](https://github.com/azure-rtos/netxduo/releases/tag/v6.2.1_rel), please try it and see if this fixes your issue.

fdesbiens, 2026-10-01:
I confirmed that v6.2.1 fixed the TLS 1.3 packet leak and that the fix still works on current `dev`. I opened #465 for the missing regression test as follow-up cleanup. I’m closing this issue as resolved.

closed 2026-10-01 by fdesbiens