eclipse-threadx/netxduo #144: Documentation gap for SNTP module broad cast mode (Server configuration)

ranganath25, 2023-01-23:
**Describe the bug**
Documentation for implementation of broadcast mode needs to be provided since it requires information on server configuration as broad cast. 
Currently, In Azure documentation, there is no much information provided on broad cast mode server side configuration which is important for user. 

**Please also mention any information which could help others to understand
the problem you're facing:**
- **What target device are you using?**
- Renesas e² studio, EK-RA6M3
**- Which version of Azure RTOS?**
- v6.2.0+fsp.4.3.0.beta.0
**- What toolchain and environment?**
- GCC - 10.3.1, Renesas e² studio, 
**- What have you tried to diagnose or workaround this issue?**
1. We tried to make linux as NTP server for broad cast mode by installing NTP on it. We changed required configuration files (ntp.conf) which is under /etc/ . Referred following link to make changes on ntp.conf file. https://kb.meinbergglobal.com/kb/time_sync/ntp/configuration/ntp_conf_file
2. Also we tried if we can get global NTP servers which has broad cast services like pool.ntp.org, time.google.com. 

Other Analysis:
1. Referred many other documents and websites from google to understand server configurations. 
3. (https://kb.meinbergglobal.com/kb/time_sync/ntp/configuration/ntp_broadcast_mode#:~:text=NTP%20Broadcast%20Server%20Configuration,-ntpd%20in%20the&text=As%20shown%20for%20the%20192.168,behind%20one%20or%20more%20routers.)
4. https://community.renesas.com/mcu-mpu/embedded-system-platform/f/forum/13837/sntp
5. https://timetoolsltd.com/ntp/how-to-install-and-configure-ntp-on-linux/

**Expected behavior**
Broad cast Server configurations information should be provided for user which is important to make SNTP to work with broad cast mode. 

Note: Here board acts as client and we need server configurations for broad cast mode.

**Impact**
Not able to test Broad cast mode. This may impact our customer who faces same issue and since it is not tested in our environment there could be field issue which we can expect from customer.

wenhui-xie, 2023-02-09:
Hi @ranganath25, thanks for your report. We will add example codes and other info for SNTP broadcast mode.

ranganath25, 2023-02-09:
Hi @wenhui-xie ,

         Thank you for your support.

wenhui-xie, 2023-11-13:
We have example codes for SNTP client with broadcast mode. We don't support SNTP server and there are no documents for SNTP server.

closed 2023-11-22 by wenhui-xie