eclipse-threadx/netxduo #147: Need help on using the iperf and network tools

panneerc, 2023-02-01:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using? We are going to deploy on Azure WVD or AVD 

Need help on using the iperf to run on Azure container and the webpage on Clients to test the Network latency .

TiejunMS, 2023-02-06:
If your device running Azure RTOS has a public IP address, then you can run both client and server test. If not, the device can run as client only. See network topology below. PC and device must be in the same network or at least PC should be able to access device. From PC browser, open http://10.0.0.2 (replace with actual IP address). Change the IP address of iperf server and start test.

[image]

panneerc, 2023-02-17:
Hello @TiejunMS  Thanks for the Information. Do we need to install client on all VM machines ?
Because i am looking for the option like running iperf/Jperf on machine from web interface

TiejunMS, 2023-02-20:
In above network topology, you only need to install iperf on Azure VM. For NetX running on device, the iperf application is already there.

closed 2023-03-09 by TiejunMS