eclipse-threadx/netxduo #148: Include reordering in 6.2.0 causes Azure IoT addon to compile with the security module enabled

renesas-adam-benson, 2023-02-02:
Renesas FSP is using `nx_user.h` to store user configuration for NetX Duo and bundled addons. This also includes defining `NX_AZURE_DISABLE_IOT_SECURITY_MODULE` in `nx_user.h` to disable the IoT security module. 

There was an include order change introduced in v6.2.0 for `nx_azure_iot.c` that breaks this behavior because `nx_azure_iot.h` is now included after `nx_azure_iot_security_module.h`:

https://github.com/azure-rtos/netxduo/blob/aad43453310a1668029bd69d8edbce8198fbca59/addons/azure_iot/nx_azure_iot.c#L17-L21

Can the `nx_azure_iot.h` be moved back above the `nx_azure_iot_security_module.h` include?

bo-ms, 2023-02-06:
Hi @renesas-adam-benson Thanks for reporting this issue, we will check it.

bo-ms, 2023-05-25:
Closing, fixed it in this commit: https://github.com/azure-rtos/netxduo/commit/939bae5c4dafde45988d593285f444702ed9be0c#diff-6cd961d99900ae5c4a4dce497351420f8508ffe2deba90ebc376ec845686cbf9

closed 2023-05-25 by bo-ms