eclipse-threadx/netxduo #150: Ask about PPP Client

Huafengland, 2023-02-07:
Ask about PPP Client
--------------------------------------------------------------------------------------
I'm tring to access network throught PPP, but it is always show PPP link down.
The tick is 1ms, so I have changed NX_IP_PERIODIC_RATE to 1000.
Here is the log I get, do you know the probable cause?
```
I/CellularNetwork [5002] InitCellularModule
I/CellularNetwork [12970] AT - RDY
I/CellularNetwork [12976] AT - AT
I/CellularNetwork [16145] AT - AT+CPIN
I/CellularNetwork [21342] AT - AT+CREG
I/CellularNetwork [21351] AT - AT+CGREG
I/CellularNetwork [21629] AT - AT+CGDCONT
I/CellularNetwork [21637] AT - ATD*99#
I/CellularNetwork [21640] InitPPPos
Time: 21642, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 1, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 8b 00 08 01 04 05 dc 
Time: 21670, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 26, Packet: c0 21 01 01 00 18 02 06 00 00 00 00 03 04 c0 23 05 06 5a 9e c9 73 07 02 08 02 
Time: 21693, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 04 01 00 08 03 04 c0 23 
Time: 21712, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 10, Packet: c0 21 02 8b 00 08 01 04 05 dc 
Time: 21730, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 3, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 22, Packet: c0 21 01 02 00 14 02 06 00 00 00 00 05 06 5a 9e c9 73 07 02 08 02 
Time: 21752, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 22, Packet: c0 21 02 02 00 14 02 06 00 00 00 00 05 06 5a 9e c9 73 07 02 08 02 
Time: 21774, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 1, Authenticated, Send Packet Length: 24, Packet: 80 21 01 8c 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00 
Time: 21806, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 03 00 04 
Time: 21823, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 22, Packet: 03 8c 00 16 03 06 64 7d 6f e4 81 06 b7 e6 7e e1 83 06 b7 e6 7e e0 
Time: 25642, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Send Packet Length: 24, Packet: 80 21 01 8d 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00 
Time: 25674, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 04 00 04 
Time: 25690, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 22, Packet: 03 8d 00 16 03 06 64 7d 6f e4 81 06 b7 e6 7e e1 83 06 b7 e6 7e e0 
<Duplicate log>
Time: 97642, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Send Packet Length: 24, Packet: 80 21 01 9f 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00 
Time: 97674, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 16 00 04 
Time: 97691, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 22, Packet: 03 9f 00 16 03 06 64 7d 6f e4 81 06 b7 e6 7e e1 83 06 b7 e6 7e e0 
I/CellularNetwork [101644] Link Down PPP State 7
Time: 101645, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 1, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 a0 00 08 01 04 05 dc 
Time: 101673, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 20, Packet: 01 17 00 14 02 06 00 00 00 00 05 06 5a 9e c9 73 07 02 08 02 
Time: 101694, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 8, Packet: 02 a0 00 08 01 04 05 dc 
Time: 105642, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 a1 00 08 01 04 05 dc 
Time: 105670, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 20, Packet: 01 18 00 14 02 06 00 00 00 00 05 06 5a 9e c9 73 07 02 08 02 
Time: 105691, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 8, Packet: 02 a1 00 08 01 04 05 dc 
<Duplicate log>
Time: 129642, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 a7 00 08 01 04 05 dc 
Time: 129670, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 20, Packet: 01 1e 00 14 02 06 00 00 00 00 05 06 5a 9e c9 73 07 02 08 02 
Time: 129691, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 8, Packet: 02 a7 00 08 01 04 05 dc 
I/CellularNetwork [143644] Link Down PPP State 1
```
The 4G Module I using is EC600.
The PPPos can work before with LwIP + FreeRTOS.
Here is the code now.
```C++
  memset(&mPacketPool, 0, sizeof(mPacketPool));
  mPacketPoolAddr = malloc(NX_PACKET_POOL_SIZE);
  if ((status = nx_packet_pool_create(&mPacketPool,
    (CHAR *)"MainPacketPool", NX_PACKET_PAYLOAD,
    mPacketPoolAddr, NX_PACKET_POOL_SIZE)) != NX_SUCCESS)
  {
    log_i("Packet Faild %d", status);
  }

  memset(&mPPPosInstance, 0, sizeof(mPPPosInstance));
  mPPPosStackMemory = malloc(8192);
  if ((status = nx_ppp_create(&mPPPosInstance,
    (CHAR *)"PPPInstance", &mIpInstance,
    mPPPosStackMemory, 8192, 15, &mPacketPool,
    invalid_packet_handler, ppp_serial_byte_output)) != NX_SUCCESS)
  {
    log_i("PPP Faild %d", status);
  }
  nx_ppp_ip_address_assign(&mPPPosInstance, IP_ADDRESS(0, 0, 0, 0), IP_ADDRESS(0, 0, 0, 0));
  nx_ppp_link_up_notify(&mPPPosInstance, link_up_callback);
  nx_ppp_link_down_notify(&mPPPosInstance, link_down_callback);

  memset(&mIpInstance, 0, sizeof(mIpInstance));
  mIpStackMemory = malloc(8192);
  if ((status = nx_ip_create(&mIpInstance,
    (CHAR *)"IPInstance", IP_ADDRESS(0, 0, 0, 0),
    0xFFFFF000UL, &mPacketPool, nx_ppp_driver,
    mIpStackMemory, 8192, 15)) != NX_SUCCESS)
  {
    log_i("IP Faild %d", status);
  }
```

bhnaphade, 2023-02-07:
Hi @Huafengland 
Before initiating call for ATD*99 , 
is the correct APN and PDP type configured using AT+CGDCONT.  Check if the apn requires username and password.

Also here is a [reference example using Netx PPP](https://learn.microsoft.com/en-us/azure/rtos/netx-duo/netx-duo-ppp/chapter2#:~:text=Small%20Example%20System)

We are working on [Embedded Wireless Framework](https://github.com/Azure/embedded-wireless-framework) and have recently added PPP example using few cellular modems. You can refer the details here
https://github.com/Azure/embedded-wireless-framework/blob/main/examples/WIN32/README_PPP.md

Huafengland, 2023-02-07:
Hello @bhnaphade 
4G Module has been configured using the following commands before ATD*99#
```
SEND: "AT\r\n"
RECV: "AT\r\r\nOK\r\n"
SEND: "AT+CPIN?\r\n"
RECV: "AT+CPIN?\r\r\n+CPIN: READY\r\n\r\nOK\r\n"
SEND: "AT+CREG?\r\n"
RECV: "AT+CREG?\r\r\n+CREG: 0,1\r\n\r\nOK\r\n"
SEND: "AT+CGREG?\r\n"
RECV: "AT+CGREG?\r\r\n+CGREG: 0,1\r\n\r\nOK\r\n"
SEND: "AT+CGDCONT=1,\"IP\",\"CMIOT\"\r\n"
RECV: "AT+CGDCONT=1,\"IP\",\"CMIOT\"\r\r\nOK\r\n"
SEND: "ATD*99#\r\n"
RECV: "ATD*99#\r\r\nCONNECT\r\n"
```
And while it was using LwIP,
I did not set username or password,
It could connect correctly.

bo-ms, 2023-02-07:
Hi @Huafengland From the log, No.1-No.6 is to negotiate LCP and complete, No.7 (Packet: **80 21** 01 ...): device sent out IPCP request to modem, but No.8 (Packet: **01 03** 00 04 ) and No.9 (Packet: **03 8c** 00) are incorrect from modem since IPCP packet should start with 80 21.
```
Time: 21806, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 03 00 04 
Time: 21823, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 22, Packet: 03 8c 00 16 03 06 64 7d 6f e4 81 06 b7 e6 7e e1 83 06 b7 e6 7e e0 
```

Huafengland, 2023-02-07:
Hello @bo-ms 

我将串口信息打印了出来
```
void CellularNetwork::RecvInnerCallback(uint8_t * buff, size_t size)
{
  elog_hexdump("Pack", 32, buff, size);
  for (uint16_t i = 0; i < size; i++)
  {
    nx_ppp_byte_receive(&mPPPosInstance, buff[i]);
  }
}
```
以下是我打印的内容
```
I/CellularNetwork [5001] InitCellularModule
I/CellularNetwork [12988] AT - RDY
I/CellularNetwork [12992] AT - AT
I/CellularNetwork [15002] AT - AT+CPIN
I/CellularNetwork [24134] AT - AT+CREG
I/CellularNetwork [24140] AT - AT+CGREG
I/CellularNetwork [24281] AT - AT+CGDCONT
I/CellularNetwork [24286] AT - ATD*99#
I/CellularNetwork [24287] InitPPPos
Time: 24293, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 1, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 e5 00 08 01 04 05 dc 
D/HEX Pack: 0000-001F: 7E FF 7D 23 C0 21 7D 21  7D 21 7D 20 7D 38 7D 22  7D 26 7D 20 7D 20 7D 20  7D 20 7D 23 7D 24 C0 23    ~.}#.!}!}!} }8}"}&} } } } }#}$.#
D/HEX Pack: 0020-003F: 7D 25 7D 26 33 9D 7D 2A  65 7D 27 7D 22 7D 28 7D  22 F2 EF 7E 7E FF 7D 23  C0 21 7D 22 E5 7D 20 7D    }%}&3.}*e}'}"}(}"..~~.}#.!}".} }
D/HEX Pack: 0040-005F: 28 7D 21 7D 24 7D 25 DC  6F 7D 26 7E                                                                  (}!}$}%.o}&~
Time: 24322, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 26, Packet: c0 21 01 01 00 18 02 06 00 00 00 00 03 04 c0 23 05 06 33 9d 0a 65 07 02 08 02 
Time: 24345, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 04 01 00 08 03 04 c0 23 
Time: 24364, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 10, Packet: c0 21 02 e5 00 08 01 04 05 dc 
D/HEX Pack: 0000-001F: 7E FF 7D 23 C0 21 7D 21  7D 22 7D 20 7D 34 7D 22  7D 26 7D 20 7D 20 7D 20  7D 20 7D 25 7D 26 33 9D    ~.}#.!}!}"} }4}"}&} } } } }%}&3.
D/HEX Pack: 0020-003F: 7D 2A 65 7D 27 7D 22 7D  28 7D 22 CF 78 7E                                                            }*e}'}"}(}".x~
Time: 24384, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 3, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 22, Packet: c0 21 01 02 00 14 02 06 00 00 00 00 05 06 33 9d 0a 65 07 02 08 02 
Time: 24405, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 22, Packet: c0 21 02 02 00 14 02 06 00 00 00 00 05 06 33 9d 0a 65 07 02 08 02 
Time: 24427, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 1, Authenticated, Send Packet Length: 24, Packet: 80 21 01 e6 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00 
D/HEX Pack: 0000-001F: 7E 80 21 01 03 00 04 03  2C 7E 7E 80 21 03 E6 00  16 03 06 64 75 85 60 81  06 78 C4 A5 07 83 06 DD    ~.!.....,~~.!......du.`..x......
D/HEX Pack: 0020-003F: B3 26 07 32 CB 7E                                                                                     .&.2.~
Time: 24461, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 03 00 04 
Time: 24478, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 22, Packet: 03 e6 00 16 03 06 64 75 85 60 81 06 78 c4 a5 07 83 06 dd b3 26 07 
Time: 28293, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Send Packet Length: 24, Packet: 80 21 01 e7 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00 
D/HEX Pack: 0000-001F: 7E 80 21 01 04 00 04 06  A0 7E 7E 80 21 03 E7 00  16 03 06 64 75 85 60 81  06 78 C4 A5 07 83 06 DD    ~.!......~~.!......du.`..x......
D/HEX Pack: 0020-003F: B3 26 07 5E FC 7E                                                                                     .&.^.~
Time: 28325, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 04 00 04 
Time: 28342, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 22, Packet: 03 e7 00 16 03 06 64 75 85 60 81 06 78 c4 a5 07 83 06 dd b3 26 07 
<Duplicate Log>
Time: 100293, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Send Packet Length: 24, Packet: 80 21 01 f9 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00 
D/HEX Pack: 0000-001F: 7E 80 21 01 16 00 04 2B  90 7E 7E 80 21 03 F9 00  16 03 06 64 75 85 60 81  06 78 C4 A5 07 83 06 DD    ~.!....+.~~.!......du.`..x......
D/HEX Pack: 0020-003F: B3 26 07 B4 92 7E                                                                                     .&...~
Time: 100326, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated, Received Packet Length: 4, Packet: 01 16 00 04 
[... cut ...]
Time: 180321, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 20, Packet: 01 2a 00 14 02 06 00 00 00 00 05 06 33 9d 0a 65 07 02 08 02 
Time: 180342, PPP Name: PPPInstance, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 8, Packet: 02 0d 00 08 01 04 05 dc 
I/CellularNetwork [184264] Link Down: 1
```
从Log查看，实际上接收到的信息中包含 80 21
但是PPP识别的包中不包含 80 21

Huafengland, 2023-02-08:
Could you help me find the cause?

It seems that ```7E 80 21 01 03 00 04 03 2C 7E``` was received from UART,
But PPP only recognized ```01 03 00 04``` at Time: 24461.

And the IPCP is always be NX_PPP_IPCP_CONFIGURE_REQUEST_SENT_STATE.

jackyzcx, 2023-02-09:
@Huafengland Hi Huafengland, This is Jacky, working on partner engagement for EWF. I'd like to know more of your needs. Do you mind to contact me by email? jacky.zhao@microsoft.com. Thanks.

Huafengland, 2023-02-09:
Hi @jackyzcx 

My problem is PPP Instance is always in the state of NX_PPP_IPCP_CONFIGURE_REQUEST_SENT_STATE.

I just found that the IPCP Configure Request
```80 21 01 e6 00 16 03 06 00 00 00 00 81 06 00 00 00 00 83 06 00 00 00 00```
contains DNS options, but DNS was not set, and don't know if that's the problem.
So I will try to set DNS and try again.

If it still doesn't work, please take a look at this problem, Thanks.

wenhui-xie, 2023-02-09:
Hi @Huafengland, we can see from the received data that the data is compressed:
`7E 80 21 01 03 00 04 03  2C 7E 7E 80 21 03 E6 00  16 03 06 64 75 85 60 81  06 78 C4 A5 07 83 06 DD`.
The data was started with "7E" not "7E FF 03", which means the Address and Control fields are compressed.

This is a known issue that the modem may send compressed data when our NetX PPP didn't send the ACFC option. You can refer to this issue https://github.com/azure-rtos/netxduo/issues/100 which is the same issue about the ACFC option. This issue is caused by the wrong behavior of the modem and our NetX PPP doesn't support the compression option.

Below is the definition of ACFC from the RFC:
[6.6](https://datatracker.ietf.org/doc/html/rfc1661#section-6.6). Address-and-Control-Field-Compression (ACFC)

```
  This Configuration Option provides a method to negotiate the
  compression of the Data Link Layer Address and Control fields.  By
  default, all implementations MUST transmit frames with Address and
  Control fields appropriate to the link framing.

  Since these fields usually have constant values for point-to-point
  links, they are easily compressed.  This Configuration Option is
  sent to inform the peer that the implementation can receive
  compressed Address and Control fields.

  If a compressed frame is received when Address-and-Control-Field-
  Compression has not been negotiated, the implementation MAY
  silently discard the frame.

  The Address and Control fields MUST NOT be compressed when sending
  any LCP packet.  This rule guarantees unambiguous recognition of
  LCP packets.

  When the Address and Control fields are compressed, the Data Link
  Layer FCS field is calculated on the compressed frame, not the
  original uncompressed frame.
```

According to RFC, this Configuration Option is sent to inform the peer that the implementation can receive such compressed fields. If our NetX PPP did not send this option to the peer, the peer should not send compressed data to NetX PPP.

jackyzcx, 2023-02-09:
@Huafengland Hope Wenhui's comment can help on your issue. I want to know more about how EWF and Azure RTOS can help your project. Please contact with me through email. I can't find your contact info on GitHub. Thanks.

Huafengland, 2023-02-09:
@wenhui-xie Thanks, I am contacting with 4G Module Manufacturer about this.
@jackyzcx I will contact you through email.

You can close this issue.

Thanks for support!

Huafengland, 2023-02-10:
After look inside the PPP protocol, I think the PPP stack of NetxDuo has a bug.
```
1. MCU -> MODULE: LCP_CONFIGURE_REQUEST
> c0 21 01 d2 00 08 01 04 05 dc
  config Maximum-Receive-Unit to 1500

2. MODULE -> MCU: LCP_CONFIGURE_REQUEST
< c0 21 01 01 00 18 02 06 00 00 00 00 03 04 c0 23 05 06 49 7b 6c d4 07 02 08 02
  config 02(unknow)
  config Authentication-Protocol to PAP
  config Magic-Number
  config Protocol-Field-Compression
  config Address-and-Control-Field-Compression

3. MCU -> MODULE: LCP_CONFIGURE_REJECT for Message 2
> c0 21 04 01 00 08 03 04 c0 23
  reject config Authentication-Protocol to PAP

4. MODULE -> MCU: LCP_CONFIGURE_ACK for Message 1
< c0 21 02 d2 00 08 01 04 05 dc
  ack config Maximum-Receive-Unit to 1500

5. MODULE -> MCU: LCP_CONFIGURE_REQUEST
< c0 21 01 02 00 14 02 06 00 00 00 00 05 06 49 7b 6c d4 07 02 08 02
  config 02(unknow)
  config Magic-Number
  config Protocol-Field-Compression
  config Address-and-Control-Field-Compression

6. MCU -> MODULE: LCP_CONFIGURE_ACK for Message 5
> c0 21 02 02 00 14 02 06 00 00 00 00 05 06 49 7b 6c d4 07 02 08 02
  ack config 02(unknow)
  ack config Magic-Number
  ack config Protocol-Field-Compression
  ack config Address-and-Control-Field-Compression

7. MCU -> MODULE: IPCP_CONFIGURE_REQUEST
> 80 21 01 d3 00 16 03 06 00 00 00 00 81 06 72 72 72 72 83 06 72 72 73 73
```
If the PPP stack does not support Address-and-Control-Field-Compression,
it should reject config Address-and-Control-Field-Compression at Message 3,
but it ack config Address-and-Control-Field-Compression at Message 6.

wenhui-xie, 2023-02-10:
Hi @Huafengland, as the RFS says, if the peer sends Address-and-Control-Field-Compression, it means it can receive the compressed data, not means it will send the compressed data. The ack from NetX PPP doesn't mean our PPP stack can process compressed data. Only if NetX PPP sends this option to the peer then the peer can send compressed data.

dimavhd, 2023-02-17:
I had a similar problem but I use another modem.
In my case the cause was that I did not properly check the network registration status:
AT+CREG?
As soon as I added a check to make sure the modem is connected to the network, everything worked.

Huafengland, 2023-02-20:
@dimavhd Thanks, I have checked this in my sequence.
This is my AT communications:
```
SEND: "AT\r\n"
RECV: "AT\r\r\nOK\r\n"
SEND: "AT+CPIN?\r\n"
RECV: "AT+CPIN?\r\r\n+CPIN: READY\r\n\r\nOK\r\n"
SEND: "AT+CREG?\r\n"
RECV: "AT+CREG?\r\r\n+CREG: 0,1\r\n\r\nOK\r\n"
SEND: "AT+CGREG?\r\n"
RECV: "AT+CGREG?\r\r\n+CGREG: 0,1\r\n\r\nOK\r\n"
SEND: "AT+CGDCONT=1,\"IP\",\"CMIOT\"\r\n"
RECV: "AT+CGDCONT=1,\"IP\",\"CMIOT\"\r\r\nOK\r\n"
SEND: "ATD*99#\r\n"
RECV: "ATD*99#\r\r\nCONNECT\r\n"
```
I have fix this problem by delete `nx_ppp.c#L3638` and `nx_ppp.c#L3639`
It should be the problem about the module which I using.

closed 2023-03-14 by bo-ms

wenhui-xie, 2023-04-21:
Hi @Huafengland , we have supported processing compressed PF and ACF. This is the latest update link of PPP: https://github.com/azure-rtos/netxduo/commit/e67c70ef358e36887bdb2ff7029807eaf234e97f. 
It would be much appreciated if you could help to test the latest code to check whether the compression issue you reported is fixed.

Below are the steps to test the new feature of PPP:
1. Replace the nx_ppp.c and nx_ppp.h in your project with the latest files: https://github.com/azure-rtos/netxduo/tree/master/addons/ppp.
2. Enable processing compressed PF and ACF by adding the macro 'NX_PPP_COMPRESSION_ENABLE' to nx_user.h or to the project configuration.
3. Rebuild and run the project.

Huafengland, 2023-08-10:
@wenhui-xie Thank you for informing me.