eclipse-threadx/netxduo #155: Use outside of an embedded system?

maxkunes, 2023-02-16:
Hi!

As part of an effort to mock the hardware elements of my system such that I can test off of actual hardware, I'll need to be able to use and test Netxduo in standard environments like compiling through MSVC and running on  x86-64 under Windows.

I've noticed ports for this environment and others. However, how could a driver be written to perform the network capabilities? My best guess is modifying the RAM driver (nx_ram_network_driver.c) to do the actual system calls. However, my worry is this driver is too low-level to do that. 

Is there a recommended way to get Netxduo running outside an embedded system in a standard operating system like Windows running as a user-mode x86 application? And by running, I mean able to send and receive network traffic.

TiejunMS, 2023-02-16:
@maxkunes , if you are interested in running NetX Duo on Linux, here is a tutorial: https://learn.microsoft.com/en-us/training/modules/build-a-udp-echo-client/3-building-and-running-the-project

And a simulator driver for Linux can be found here: https://github.com/Azure-Samples/azure-rtos-learn-samples/blob/main/courses/netxduo/Driver/nx_linux_network_driver.c

maxkunes, 2023-02-16:
Awesome, that looks like what I would need. Porting that to windows seems
possible with winpcap.

It would be nice to have some of these samples in the main repo. Didn't
even realize that the other repo existed.

Thanks!

hwmaier, 2023-02-16:
> Awesome, that looks like what I would need. Porting that to windows seems possible with winpcap.

I have done a Windows port using winpcap.  But the code is not published and only used internally as NetXDuo is not really open source and I am uncertain of the license terms.

closed 2023-03-09 by TiejunMS