eclipse-threadx/netxduo #156: About RSA root key size for Azure ADU manifest verification.

ReKawashima, 2023-02-21:
The step 2.4 in nx_azure_iot_adu_agent_manifest_verify() verify the update manifest received from Azure ADU side, and it requires the RSA root key hardcorded in agent device. 
The Azure NetX Duo IoT sample code provides production and test keys in [nx_azure_iot_adu_root_key.c](https://github.com/azure-rtos/netxduo/blob/master/addons/azure_iot/nx_azure_iot_adu_root_key.c), however, its n size is 385 byte, not 384. In the [Azure ADU sample code](https://github.com/azure-rtos/netxduo/blob/master/addons/azure_iot/nx_azure_iot_adu_agent.c#L3677), the RSA temporary buffer and some parameters are set as 384 bytes size, which does not match the size of sample keys. And it makes crypto operation errors. Could you check if the root keys data (and size) is correct?
Perhaps 0x00 at the beginning of the data is unnecessary. I assume that the 0x00 was added according to ASN.1 coding, but I'm not sure if the NetX Crypto _nx_crypto_method_rsa_operation() API expects keys in ASN.1 format data.

ReKawashima, 2023-03-16:
Hi @bo-ms thanks for the comments. Yes, I will try again with the latest version of netx-duo. 

>BTW, could you share the issue you encountered?

I remember that the program stopped in exception handler due to a stack overflow, when decording the ADU manifest. I will reconfirm this as well.

bo-ms, 2023-03-17:
Hi @ReKawashima Similar issue, ADU agent calls RSA to verify the manifest, unalignment in RSA operation may cause hardfault on some devices.

closed 2023-09-20 by bo-ms