eclipse-threadx/netxduo #160: dns请求失败，会造成内存泄漏

chengshuihang, 2023-03-21:
**Describe the bug**
当我使用nxd_dns_host_by_name_get获取ip时，发生失败事件后，nx内存池nx_packet_pool_available数量减少，不会恢复

Please also mention any information which could help others to understand
the problem you're facing:
- 使用nxd_dns_host_by_name_get且返回失败
- netxduo软件版本v6.2.0
- STM32CubeIDE 1.12.0
- 修改_nx_dns_response_get函数，增加nx_packet_release调用

[image]

bo-ms, 2023-03-21:
@chengshuihang thanks for reporting the issue, I did a quick check for[ _nx_dns_reponse_get () ](https://github.com/azure-rtos/netxduo/blob/v6.2.0_rel/addons/dns/nxd_dns.c#L4652)function, looks all functions in _nx_dns_response_get() correctly release the packet for error status, could you point us where we need to release the packet? thanks.

```
UINT _nx_dns_response_get(NX_DNS *dns_ptr, UCHAR *host_name, UCHAR *record_buffer, 
                          UINT buffer_size, UINT *record_count, ULONG wait_option)
{
UINT        status;
NX_PACKET  *packet_ptr;

    /* Wait for a DNS response.  */
    status = _nx_dns_response_receive(dns_ptr, &packet_ptr, wait_option);

    /* Check status.  */
    if (status == NX_SUCCESS)
    {

#ifndef NX_DISABLE_PACKET_CHAIN
        if (packet_ptr -> nx_packet_next)
        {
            
            /* Chained packet is not supported. */
            nx_packet_release(packet_ptr);

            /* Release the resource obtained in _nx_dns_host_resource_data_by_name_get for non-blocking.  */
            if (wait_option == NX_NO_WAIT)
            {

                /* Unbind the socket.  */
                nx_udp_socket_unbind(&(dns_ptr -> nx_dns_socket));
                tx_mutex_put(&dns_ptr -> nx_dns_mutex);
            }

            return(NX_INVALID_PACKET);
        }
#endif /* NX_DISABLE_PACKET_CHAIN */

        /* Call the function to process the DNS packet.  */
        status = _nx_dns_response_process(dns_ptr, host_name, packet_ptr, record_buffer, buffer_size, record_count);
    }

    /* Release the resource obtained in _nx_dns_host_resource_data_by_name_get for non-blocking.  */
    if (wait_option == NX_NO_WAIT)
    {

        /* Unbind the socket.  */
        nx_udp_socket_unbind(&(dns_ptr -> nx_dns_socket));
        tx_mutex_put(&dns_ptr -> nx_dns_mutex);
    }

    return(status);
}
```

chengshuihang, 2023-03-21:
对不起，你是对的。我问题方向找错了，应该是nxd_udp_socket_send正确返回，却没有释放包。
[image]
[image]
nxd_udp_socket_send返回NX_SUCCESS，但是nx_packet_pool_available却没有增加

chengshuihang, 2023-03-21:
nxd_udp_socket_send发送成功的包，应该哪里被销毁？

bo-ms, 2023-03-21:
@chengshuihang NX_SUCCESS means the packet has been sent to network driver, then driver should release the packet after transmission, refer to https://learn.microsoft.com/en-us/azure/rtos/netx-duo/chapter4#nxd_udp_socket_send

chengshuihang, 2023-03-21:
ok，你关闭它吧，等我找到原因，我再开

closed 2023-03-21 by bo-ms