eclipse-threadx/netxduo #161: Supporting RSA signed client certificates with TLS 1.3

jkuwaha, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
With TLS 1.3, TLS session cannot be established with clients using RSA signed certificates.

**Describe the solution you'd like**
It would be good if NetX supports RSA signed client certificates with TLS 1.3.

TiejunMS, 2023-03-23:
@jkuwaha , thanks for opening this feature request. This feature is in our backlog. It is not prioritized at the moment.

jkuwaha, 2023-03-23:
@TiejunMS,
Thank you for the information.
We look forward to being supported (hopefully within a year).

GKNSTM, 2024-06-11:
Hello. Any news on this feature request ? we would be very interested to have this RSA certificate support in TLS 1.3 in Netxduo.