eclipse-threadx/netxduo #162: fin packet loss problem

rk3399, 2023-03-28:
When netxduo acts as a tcp host（192.168.0.101）, the tcp client（192.168.0.100） sends fin packets, and netxduo only responds to ack and does not send fin packets
[image]

TiejunMS, 2023-03-29:
@rk3399 , could you double check if nx_tcp_socket_disconnect is invoked when FIN is received from peer socket?

rk3399, 2023-03-29:
> @rk3399 , could you double check if nx_tcp_socket_disconnect is invoked when FIN is received from peer socket?
Sorry, I don't know how to do this. I thought netxduo would handle this process automatically. Is nx_tcp_socket_disconnect called in the last parameter disconnect_callback of nx_tcp_socket_create?

rk3399, 2023-03-29:
还有一个疑问。即使在nx_user.h中定义了NX_ENABLE_TCP_KEEPALIVE，netxduo作为tcp服务器时，也并没有向客户端发送keepalive

TiejunMS, 2023-03-29:
You can invoke nx_tcp_socket_disconnect in disconnect_callback function, which is in IP thread context. Since it is not allowed to block in IP thread, the wait option specified in nx_tcp_socket_disconnect will not work as expected. If your application is doing blocking call on nx_tcp_socket_receive, it will return NX_NOT_CONNECTED when FIN is received. Then you can invoke nx_tcp_socket_disconnect to close the connection.

For keepalive feature, you will need to pay attention to following [options](https://learn.microsoft.com/en-us/azure/rtos/netx-duo/chapter2#configuration-options).
NX_TCP_KEEPALIVE_INITIAL
NX_TCP_KEEPALIVE_RETRIES
NX_TCP_KEEPALIVE_RETRY

By default, the first keepalive packet will be sent after 2 hours.

rk3399, 2023-03-29:
这很有效，感谢您的解答，暂时没有别的问题了。我将关闭此issuse

closed 2023-03-29 by rk3399