eclipse-threadx/netxduo #163: 不能完整发送nx_packet的问题

rk3399, 2023-03-30:
该问题出现在stm32h7平台，开启数据包分片。发送一个巨大的数据包，被分成超过ETH_TX_DESC_CNT（默认为4）,第4包之后的数据包不会被发送。跟踪发现是因为发送数据包的线程一直占有NX _IP->nx_ip_protection。使得IP线程一直被挂起，不能及时释放已经发送的前4个数据包。我想知道有什么解决方法吗，谢谢
[image]

rk3399, 2023-03-30:
好的好的，我已经找到解决办法了，如果有需要可以分享给其他人
1、将_nx_driver_deferred_processing中的HAL_ETH_ReleaseTxPacket调用剪切到HAL_ETH_TxCpltCallback
2、_nx_driver_hardware_packet_send中循环等待HAL_ETH_Transmit_IT返回HAL_OK

现在关闭

TiejunMS, 2023-03-30:
Good to know you have solved your problem and thanks for sharing your solution! For such a workaround, the performance of TCP/IP stack would be impacted due to the transmission of each packet is blocked until sent out. Ideally, the number of TX BD should be larger than the total packets from pool to avoid unexpected packet drop.

rk3399, 2023-03-30:
我的理解是如果分片的数量小于ETH_TX_DESC_CNT，性能是不受影响的,因为HAL_ETH_Transmit_IT总是返回HAL_OK。当应用程序有大量的数据包要发送，而ETH发送的速度不足，那么应当阻塞应用程序等待ETH就绪。否则就会产生丢包

TiejunMS, 2023-03-30:
The performance impact is not only for IP fragments, but also for all TCP/IP packets. If you run the iPerf test, the benchmark will be largely impacted. But your analysis of packet drop is correct.

rk3399, 2023-03-30:
非常感谢，我会深入分析这一处理产生的影响