eclipse-threadx/netxduo #165: Wrong caller checking in nx_dhcp_state_change_notify

OortJacek, 2023-04-03:
https://github.com/azure-rtos/netxduo/blob/939bae5c4dafde45988d593285f444702ed9be0c/addons/dhcp/nxd_dhcp_client.c#L3602

Cannot set the DHCP's notify function from the initialization. The documentation says that it is allowed to call this function from the initialization. Furthermore, all other functions that set notify functions are allowed from the initialization. 

It should be: NX_INIT_AND_THREADS_CALLER_CHECKING

TrumsZ, 2023-04-04:
Hello, thank you for your suggestion. We will take a look at it.

bo-ms, 2023-05-25:
Closing. The caller checking has been corrected in this commit: https://github.com/azure-rtos/netxduo/commit/5b5c8c580684cd2f85001ff1763a62e0fdac9452#diff-474e87983b8ff48b783265e25bf5b6f5320d2460339a1bd29e22ae3e452b1cf8

closed 2023-05-25 by bo-ms