eclipse-threadx/netxduo #166: Failing sending SMTP using Microsoft Acount. Device Used: EVK-MIMXRT1060

lawrencerb, 2023-04-07:
Is this request related to a particular hardware platform, SoC, board? Please describe.
I am working with the NXP MIMXRT1060-EVK board, which uses an i.MX RT1060 crossover MCU. I am currently using Azure RTOS NetX Duo library for my project, specifically focusing on the SMTP client to send emails through an Outlook account (smtp.office365.com).

Describe why you are asking for this support?
While implementing the SMTP client, I am encountering an authentication error (NX_SMTP_AUTH_REPLY_ERROR) when attempting to send an email using the smtp.office365.com server. I have tried different configurations, including using an SMTP relay configuration from the Outlook documentation: https://learn.microsoft.com/en-us/exchange/mail-flow-best-practices/how-to-set-up-a-multifunction-device-or-application-to-send-email-using-microsoft-365-or-office-365. However, I am still unable to resolve the issue.

I am asking for support to help me understand the cause of this error and identify the necessary steps or modifications required to send emails using the Azure RTOS NetX Duo SMTP client with the Outlook SMTP server.

Willingness to maintain Azure RTOS support
If there are any updates or changes to the Azure RTOS support for the NXP MIMXRT1060-EVK board or the i.MX RT1060 SoC, I am willing to work with the Azure RTOS team to maintain and update the support as needed, ensuring compatibility and functionality for future projects.

Additional context
I have provided a code snippet of my current implementation in a previous post. Please let me know if any additional information or context is required for understanding the issue.

Feel free to modify and add any relevant information before submitting your request.
Modified the shared>source>networking.c to include the following:
```

/* Define the host user name and mail box parameters */
#define USERNAME               "lawrencetrivia@outlook.com" //just an email I created to test it
#define PASSWORD               "passwordremovedforprivacyissues"
#define FROM_ADDRESS           "lawrencetrivia@outlook.com"
#define RECIPIENT_ADDRESS      "lawrence.boechat@conceptu.ind.br"
#define LOCAL_DOMAIN           "mycompany.com"

#define SUBJECT_LINE           "NetX Duo SMTP Client Demo"
#define MAIL_BODY              "NetX Duo SMTP client is an SMTP client \r\n" \
                               "implementation for embedded devices to send  \r\n" \
                               "email to SMTP servers. This feature is \r\n" \
                               "intended to allow a device to send simple \r\n " \
                               "status reports using the most universal \r\n " \
                               "Internet application, email.\r\n"

/* See the NetX Duo SMTP Client User Guide for how to set the authentication type.
   The most common authentication type is PLAIN. */
#define CLIENT_AUTHENTICATION_TYPE 1

#define SMTP_SERVER            "smtp.office365.com"
#define SERVER_PORT            587 // Use port 587 for STARTTLS

#define NETX_IP_STACK_SIZE  2048
#define NETX_PACKET_COUNT   60
#define NETX_PACKET_SIZE    1536
#define NETX_POOL_SIZE      ((NETX_PACKET_SIZE + sizeof(NX_PACKET)) * NETX_PACKET_COUNT)
#define NETX_ARP_CACHE_SIZE 512
#define NETX_DNS_COUNT      6

#define NETX_IPV4_ADDRESS IP_ADDRESS(0, 0, 0, 0)
#define NETX_IPV4_MASK    IP_ADDRESS(255, 255, 255, 0)

#define DHCP_WAIT_TIME_TICKS (30 * TX_TIMER_TICKS_PER_SECOND)

static UCHAR netx_ip_stack[NETX_IP_STACK_SIZE];
static UCHAR netx_ip_pool[NETX_POOL_SIZE];
static UCHAR netx_arp_cache_area[NETX_ARP_CACHE_SIZE];

static NX_DHCP nx_dhcp_client;

NX_IP nx_ip;
NX_PACKET_POOL nx_pool;
NX_PACKET_POOL client_packet_pool;
NX_DNS nx_dns_client;
static NX_SMTP_CLIENT           demo_client;

```

Also:
```
UINT network_connect()
{
    UINT status;
    NXD_ADDRESS server_ip_address;
    UINT        error_counter = 0;

    // Fetch IP details
    if ((status = dhcp_connect()))
    {
        printf("ERROR: dhcp_connect\r\n");
    }

    // Create DNS
    else if ((status = dns_connect()))
    {
        printf("ERROR: dns_connect\r\n");
    }

    // Wait for an SNTP sync
    else if ((status = sntp_sync()))
    {
        printf("ERROR: Failed to sync SNTP time (0x%08x)\r\n", status);
    }

        tx_thread_sleep(500);

    /* Set up the server IP address. */
    server_ip_address.nxd_ip_version = NX_IP_VERSION_V4;
    server_ip_address.nxd_ip_address.v4 = 0;

    // Resolve SMTP server's IP address
    if ((status = dns_resolve_smtp_server(&server_ip_address)))

    {
        printf("ERROR: Failed to resolve SMTP server's IP address (0x%08x)\r\n", status);
    }

    /* The demo client username and password is the authentication
       data used when the server attempts to authentication the client. */

    status =  nxd_smtp_client_create(&demo_client, &nx_ip, &nx_pool,
                                     USERNAME,
                                     PASSWORD,
                                     FROM_ADDRESS,
                                     LOCAL_DOMAIN, CLIENT_AUTHENTICATION_TYPE, &server_ip_address, SERVER_PORT);

    if (status != NX_SUCCESS)
    {
        printf("Error creating the mail client. Status: 0x%x.\n\r", status);
        return status;
    }
        // Enable STARTTLS
    status = nxd_smtp_client_enable_starttls(&demo_client);
    if (status != NX_SUCCESS)
    {
        printf("Error enabling STARTTLS. Status: 0x%x.\n\r", status);
        return status;
    }

    /* Create a mail instance with the above text message and recipient info. */
    status =  nx_smtp_mail_send(&demo_client, RECIPIENT_ADDRESS,
                                NX_SMTP_MAIL_PRIORITY_NORMAL,
                                SUBJECT_LINE, MAIL_BODY, sizeof(MAIL_BODY) - 1);

    /* Check for errors. */
    if (status != NX_SUCCESS)
    {

        /* Mail item was not sent. Note that we need not delete the client. The
           error status may be a failed authentication check or a broken connection.
           We can simply call nx_smtp_mail_send again.  */
        printf("Error sending the email. Status: 0x%x.\n\r", status);
    }

    /* Release resources used by client. Note that the transmit packet
       pool must be deleted by the application if it no longer has use for it.*/
    status = nx_smtp_client_delete(&demo_client);

    /* Check for errors. */
    if (status != NX_SUCCESS)
    {
        error_counter++;
        printf("Error Counter: %d \r\n", error_counter);
    }

    return status;
}
```

My Stack also Include the DNS retriver:
```

UINT dns_resolve_smtp_server(NXD_ADDRESS *server_ip_address)
{
    UINT status;
    UCHAR *server_name = (UCHAR *)SMTP_SERVER;
    ULONG wait_option = 5 * NX_IP_PERIODIC_RATE;
    UINT max_attempts = 3; // Maximum number of DNS resolution attempts
    UINT attempt_count = 0;

    printf("\nResolving SMTP server's IP address\n");

    do {
        status = nxd_dns_host_by_name_get(&nx_dns_client, server_name, server_ip_address, wait_option, NX_IP_VERSION_V4);
        attempt_count++;

        if (status != NX_SUCCESS)
        {
            printf("Failed to resolve SMTP server's IP address (0x%08x), attempt %u\n", status, attempt_count);
            tx_thread_sleep(wait_option); // Sleep before retrying
        }
    } while (status != NX_SUCCESS && attempt_count < max_attempts);

    if (status == NX_SUCCESS)
    {
        printf("SUCCESS: SMTP server's IP address resolved\n");
    }

    return status;
}
```

It works, but the device prompts:
Connecting to the network

Initializing DHCP
	MAC: 02:11:46:29:F8:65
	IP address: 192.168.137.182
	Mask: 255.255.255.0
	Gateway: 192.168.137.1
SUCCESS: DHCP initialized

Initializing DNS client
	DNS address: 192.168.137.1
SUCCESS: DNS client initialized

Initializing SNTP time sync
	SNTP server 0.pool.ntp.org
	SNTP server 1.pool.ntp.org
	SNTP server 2.pool.ntp.org
	SNTP time update: Apr 7, 2023 17:55:56.981 UTC 
SUCCESS: SNTP initialized

Resolving SMTP server's IP address
SUCCESS: SMTP server's IP address resolved
Error sending the email. Status: 0xad.

0XAD stands for: #define NX_SMTP_AUTH_REPLY_ERROR                0xAD                      /* Error in response to Client SMTP AUTH command */

Any answer is more than welcomed

TiejunMS, 2023-04-10:
@lawrencerb , just so you know, TLS is not supported by NetX SMTP.

lawrencerb, 2023-04-10:
> @lawrencerb , just so you know, TLS is not supported by NetX SMTP.

I didn't knew! I'm sorry and thank you for your reply. Do you think it's possible to implement it

closed 2023-04-10 by lawrencerb

TiejunMS, 2023-04-10:
No worries. We don't have a plan to support TLS in SMTP in the near future.