eclipse-threadx/netxduo #168: bug in utility/iperf/nx_iperf.c pointer used before set its value

xuzihan351, 2023-04-27:
[https://github.com/azure-rtos/netxduo/blob/master/utility/iperf/nx_iperf.c](url)     +118
```
    /* Create a TCP server socket.  */
    status =  nx_tcp_socket_create(nx_iperf_test_ip, &tcp_server_socket, "TCP Server Socket",
                                   NX_IP_NORMAL, NX_FRAGMENT_OKAY, NX_IP_TIME_TO_LIVE, 32 * 1024,
                                   NX_NULL, nx_iperf_tcp_rx_disconnect_received);
```
Should `nx_iperf_test_ip` be `ip_ptr`? `nx_iperf_test_ip` is a null pointer but be used before set its value.

bo-ms, 2023-04-28:
Hi @xuzihan351 Thanks for reporting the issue, we are checking it.

yajunxiaMS, 2023-08-17:
This issue has been fixed internally and will be reflected in the next release.

bo-ms, 2023-11-01:
Hi @xuzihan351 The issue has been fixed in the latest version 6.3.0, could you try it? thanks.

closed 2023-11-01 by bo-ms