eclipse-threadx/netxduo #169: Does netxduo have a function to send a IPv6 RA message

prarthana-13, 2023-04-28:
Hi,
Does the netxduo have or support a function to send a IPv6 RA message . If not how can we be able to send a RA message in response to RS message

TiejunMS, 2023-04-28:
Sending IPv6 RA is not supported by NetXDuo. To send RA message, you can use API nxd_ip_raw_packet_send() to construct RA header. To receive RS message, set a raw packet filter by nx_ip_raw_packet_filter_set() to hijack RA message.

shahab5, 2023-05-01:
HI @TiejunMS we see that DHCPv6 server https://github.com/azure-rtos/netxduo/blob/master/addons/dhcp/nxd_dhcpv6_server.c implementation is present in the NetxDuo, how does it work without the RA messages? Does the DHCPv6 implementation support Stateful DHCPv6? Please help to understand the implementation done for DHCPv6 Server in this file.

TiejunMS, 2023-05-04:
@bo-ms , could you help with this question?

bo-ms, 2023-05-04:
Hi @shahab5 DHCPv6 is stateful address autoconfiguration protocol for IPv6 ([RFC3315](https://datatracker.ietf.org/doc/html/rfc3315)), this protocol is a stateful counterpart to "IPv6 Stateless Adress Autoconfiguration (RA)" ([RFC2462](https://datatracker.ietf.org/doc/html/rfc2462)). Our DHCPv6 implementation supports stateful address autoconfiguration, DHCPv6 Client/Server uses Solicit, Advertise, Request, Confirm, etc to configure address and other information. If you want to use DHCPv6 in your network, you can refer to the sample https://github.com/azure-rtos/netxduo/blob/master/samples/demo_netxduo_dhcpv6.c

prarthana-13, 2023-05-09:
Hi @TiejunMS @bo-ms , we have two observations w.r.t  RA message and ip configuration.

1.There is no RA message sent in response for RS 
[image]

2. Even without RA message from server side, DHCPv6 handshake happens and an ip address is assigned.
[image]

Please do clarify this behaviour

bo-ms, 2023-05-10:
Hi @prarthana-13 1. Your network (Router) may not support/enable IPv6 Stateless Address configuration. 2. DHCPv6 is completely independent feature, does not depend on RA/RS.

shahab5, 2023-05-22:
@bo-ms, please refer to this page https://www.networkacademy.io/ccna/ipv6/stateful-dhcpv6
It described that  A, O, M flags of RA are required by the host to decide whether Stateless or Stateful DHCPv6 should be used.

The problem is with the NetX device when configured as DHCPv6 server, we see the Third party hosts are unable to configure the IP address, while they work fine with other DHCPv6 routers that send an RA in response to RS. We think not sending an RA is leading to IOP issues. Please help to mitigate this.

bo-ms, 2023-05-22:
Hi @shahab5 Understood, router may combine/manage these features. But NetX just supports Stateful DHCPv6, as Tiejun suggested, you may implement the logic of processing RS and sending RA in your application. Thanks.

closed 2023-05-25 by bo-ms