eclipse-threadx/netxduo #174: nxd_mdns.h refers NX_IPV6_MULTICAST_ENABLE

jkuwaha, 2023-06-02:
Hi,

nxd_mdns.h reports an error "mDNS IPv6 is not supported if IPv6 multicast is not enabled." even when build with `NX_MDNS_ENABLE_IPV6` and `NX_ENABLE_IPV6_MULTICAST`.

According to source code, nxd_mdns.h refers to `NX_IPV6_MULTICAST_ENABLE` instead of `NX_ENABLE_IPV6_MULTICAST`.

https://github.com/azure-rtos/netxduo/blob/5b5c8c580684cd2f85001ff1763a62e0fdac9452/addons/mdns/nxd_mdns.h#L133-L135

If this is not intentional, could you please use `NX_ENABLE_IPV6_MULTICAST` to make configuration simpler?

bo-ms, 2023-06-02:
Hi @jkuwaha thanks for reaching out to us and providing suggestion. Only using NX_ENABLE_IPV6_MULTICAST can make configuration simpler, but adding NX_MDNS_ENABLE_IPV6, user will have choice to disable mDNS IPv6 message to decrease the code size and save bandwidth in IPv6 network. As the configuration has been released, we will not update it. Thank you for your understanding.

jkuwaha, 2023-06-02:
Hi @bo-ms,

Sorry if the explanation is hard to understand.
NX_MDNS_ENABLE_IPV6 is not the problem.

My point is that NX_IPV6_MULTICAST_ENABLE should be replaced with NX_ENABLE_IPV6_MULTICAST.
It looks like NX_IPV6_MULTICAST_ENABLE was changed to NX_ENABLE_IPV6_MULTICAST at some point.

https://github.com/azure-rtos/netxduo/blob/5b5c8c580684cd2f85001ff1763a62e0fdac9452/common/inc/nx_api.h#L748-L753

bo-ms, 2023-06-05:
Hi @jkuwaha, sorry for the misunderstanding, you are right, we should check NX_ENABLE_IPV6_MULTICAST instead of NX_IPV6_MULTICAST_ENABLE, I will fix it. Thanks again.

bo-ms, 2023-11-01:
Hi @jkuwaha The issue has been fixed in the latest version 6.3.0, could you try it? thanks.

closed 2023-11-01 by bo-ms