eclipse-threadx/netxduo #178: Tcp dup ack number is not counted right

xuzihan351, 2023-06-14:
There was an error in calculating the number of TCP dup acks. See code in nx_tcp_socket_state_ack_check.c + 246
https://github.com/azure-rtos/netxduo/blob/5b5c8c580684cd2f85001ff1763a62e0fdac9452/common/src/nx_tcp_socket_state_ack_check.c#LL246C92-L246C92
```
if (tcp_header_ptr -> nx_tcp_acknowledgment_number == starting_tx_sequence)
```
The starting_tx_sequence 's value is assigned from the following code in the function beggening.
```
starting_tx_sequence = socket_ptr -> nx_tcp_socket_tx_sequence - socket_ptr -> nx_tcp_socket_tx_outstanding_bytes;
```
If the acknowledgment number is bigger than the starting_tx_sequence, it means that all the data before acknowledgment is recieved, but it is also possibly caused by the other side discovery of `packet loss`. In this case, the ack should be considered as the first dup ack packet. But as the code using `==`, the dup ack num will not increase.  I think it should be `>=`. The `socket_ptr -> nx_tcp_socket_tx_outstanding_bytes` is updated later in the function, but the dup ack num was processed too early.
Some tcp stack will send dup ack only 3 times, this will not cause the fast retransmition.
One wireshark log is below, describe bug above. It will cause tcp speed drastic fluctuations in tcp speed.
[image]

[image]

xuzihan351, 2023-06-14:
This is what `SACK` means. 
In feature `SACK`, the first ack will be considered as a dup ack.
So netx is not supported `SACK`?

TiejunMS, 2023-06-14:
You are right. SACK is not supported by NetXDuo.

closed 2023-06-29 by TiejunMS