eclipse-threadx/netxduo #179: BSD and Cybersecurity issues

ChanakyaMacharla, 2023-06-16:
Hi techies,

I am going through the NetX Duo documentation and source code,

1. As per the NetX Duo documentation, BSD APIs are not fully compatible, is there any future release planned to support all BSD calls
2. In source code We had observed critical and cybersecurity issues while running static code analysis tool on NetX Duo firmware. are you having same issue?

TiejunMS, 2023-06-29:
Duplicate as support ticket 2306150030004913. Close this one to unify the discussion.

closed 2023-06-29 by TiejunMS

hwmaier, 2023-07-04:
@TiejunMS Do you know when Microsoft is going to release patches or updates to address those cybersecurity issues?

TiejunMS, 2023-07-05:
@hwmaier , just so you know, we haven't received issue report yet.