eclipse-threadx/netxduo #181: Features of Functional Components and Add-ons

ChanakyaMacharla, 2023-06-16:
Hi techies,

I am going through the NetX Duo documentation, I Have few points to understand

1. I can observe Either DHCP or BOOTP client must selected in compile time with help of “NX_DHCP_BOOTP_ENABLE” macro. Is there any way to choose DHCP or BOOTP client at runtime?
2. We observed that the TCP PORT need to be defined at compile time and cannot be modified runtime. For Example: In http server “nxd_http_server.h” addon, “NX_HTTP_SERVER_PORT“ macro to be defined at compile time. is that port should be configurable in runtime?
3. It is mentioned there is no limit on socket creation as per document, we would like to understand socket memory requirement with and without security feature, is there any details? 
4. As per the NetX DUO documentation, “Multihome Support” is available. In same way is it having network switch capability support as well? Our hardware is capable for Network switch.

bo-ms, 2023-06-19:
Hi @ChanakyaMacharla Thanks for asking.
1. DHCP or BOOTP Client is selected in compile time.
2. Suggest using web addon instead of http as both HTTP and HTTPS are supported in web addon, and server port can be modified runtime in [nx_web_http_server_create()](https://github.com/azure-rtos/netxduo/blob/master/addons/web/nx_web_http_server.h#L473-L479).
3. The instance of each socket is provided by user application. [Here](https://learn.microsoft.com/en-us/azure/iot-develop/concepts-azure-rtos-security-practices) is the link to security guidance.
4. It depends on the user application. We have seen our customers deploying NetXDuo to network switches and routers. But NetXDuo is mostly used in end device.

TiejunMS, 2023-06-29:
Duplicate as support ticket 2306150030004913. Close this one to unify the discussion.

closed 2023-06-29 by TiejunMS