eclipse-threadx/netxduo #183: DHCP Server : Handshake failure with GCC 11.3

krshnarajd, 2023-06-16:
**Describe the bug**
We testing netxduo in our kit (CY8CKIT-062S2-43012) and recently we have upgraded gcc version to 11.3. During testing our kit as DHCP server, we noticed that DHCP handshake is failing. The dhcp client(PC) is getting NAK from the server. 

This issue is happening with Optimization -Os enabled and is working if I remove the optimization.

Please also mention any information which could help others to understand the problem you're facing:
- What target device are you using? - CY8CKIT-062S2-43012
- Which version of Azure RTOS? - 6.2.1
- What toolchain and environment? - Toolchain GCC, Modustoolbox environment
- What have you tried to diagnose or workaround this issue? - Yes. This is happening due the strict aliasing. While the DHCP server is parsing the client options, it  tries to get the option length by calling '_nx_dhcp_server_get_data'. But the parameter is of type UINT where it is explicitly type casted to ULONG* and given to the function. This looks to be violating the aliasing.

            /* Get the option length. */
            _nx_dhcp_server_get_data(work_ptr, 1, (ULONG *)&size);   

Here the returned size is always 0. Hence the validations fails, and server rejects the client request. I tried changing the type of size of ULONG and it worked.

- work around 
-       1. During compilation, use -fno-strict-aliasing

**To Reproduce**
Steps to reproduce the behavior:
1. Build threadx + netxduo with GCC 11.3 and Optimization -Os
2. Start DHCP server
3. Client is unable to get ip from DHCP server.

**Expected behavior**
Client is able to complete DHCP Handshake

**Impact**
Severe

**Logs and console output**

**Additional context**

bo-ms, 2023-06-19:
Hi @krshnarajd Thanks for reporting the issue, we will take a look at it and let you know.

TrumsZ, 2023-06-29:
Hello @krshnarajd , may you let us know how UINT and ULONG are type-defined in tx_port.h in your project, thanks.

krshnarajd, 2023-06-29:
/* Define ThreadX basic types for this port.  */

#define VOID                                   void
typedef char                                    CHAR;
typedef unsigned char                    UCHAR;
typedef int                                       INT;
typedef unsigned int                       UINT;
typedef long                                    LONG;
typedef unsigned long                    ULONG;
typedef unsigned long long            ULONG64;
typedef short                                   SHORT;
typedef unsigned short                   USHORT;

TrumsZ, 2023-07-14:
Hi @krshnarajd , thanks for your provided information. We reproduced the phenomenon and found the assembly codes generated by the compiler optimization calls the function _nx_dhcp_server_get_data to get the value of "size" first and then re-initialize a zero value to "size". Since the compiler behaves like this, for compatibility purpose, we improved the source code internally by changing the type of the local variable "size" from UINT to ULONG as you referred to. The improvement will be available in the next official release.

bo-ms, 2023-11-01:
Hi @krshnarajd The issue has been fixed in the latest version 6.3.0, could you try it? thanks

closed 2023-11-01 by bo-ms