eclipse-threadx/netxduo #186: TLS_1_2 cannot be activated on CubeMx

Camelito30, 2023-07-14:
Hello everyone,

for my project, I'm using the evaluation board (STM32H745), the Azure RTOS (Version 3.1.0), NetX Duo (Version 6.2.0) and the MQTT stack. I want to connect to my Broker with a secure connection using nxd_mqtt_client_secure_connect(). The problem is that secure communication uses the TLS 1.2 protocol, which unfortunately cannot be enabled in CubeMx - only versions 1.0, 1.1 and 1.3 are available.

NX_SECURE_TLS_ENABLE_TLS_1_0
NX_SECURE_TLS_ENABLE_TLS_1_1
NX_SECURE_TLS_ENABLE_TLS_1_3

But would it be possible to activate NX_SECURE_TLS_ENABLE_TLS_1_2 manually in the source code, and if so, how?

Thanks in advance and I would appreciate answers or any assistance

yanwucai, 2023-07-17:
TLS 1.2 is enabled by default, and you don't need to define NX_SECURE_TLS_ENABLE_TLS_1_2 manually.

Camelito30, 2023-07-17:
Thank you for your reply. But how can I be sure that the TLS_1_2 version is defined by default? because after generating my programm I always see TLS_1_3 , even though it is not active in CubeMx.

Camelito30, 2023-07-17:
What elements do I need to enable for secure communication to work, because in your documentation it simply states that the NX_SECURE_ENABLE must be active in the Mqtt client, but what about the TLS parameters themselves?

yanwucai, 2023-07-18:
Actually TLS 1.2 cannot be disabled in the current version. The source files containing tls_1_3 in their names are always included in the project but the code is not enabled when NX_SECURE_TLS_ENABLE_TLS_1_3 is not defined.

According to the last section in [this documentation](https://learn.microsoft.com/en-us/azure/rtos/netx-duo/netx-duo-mqtt/chapter1), you need to provide a TLS setup callback function. And in this function, you shall initialize TLS module, create a TLS session, configure security parameters, and load relevant certificates to be used during TLS handshake.

You can find more information on configuring security parameters in [NetX Secure TLS user guide](https://learn.microsoft.com/en-us/azure/rtos/netx-duo/netx-secure-tls/chapter1).

TiejunMS, 2023-12-28:
Closing inactive issues.

closed 2023-12-28 by TiejunMS