eclipse-threadx/netxduo #188: Netxduo is "blocked" after an error occured

ZerAtaii, 2023-08-08:
- What target device are you using?
LPC55S69
- Which version of Azure RTOS?
6.1
- What toolchain and environment?
arm-none-eabi-* + WSL

Hi,
I use Azure RTOS with the netxduo layer to manage TCP/IP sockets. The application exchanges a set of commands through netcat via "Windows sub Linux" on different ports (32071 to 32076) with an ethernet over USB protocol.
In my application, there are different communication buses, wifi / BLE / cellular network / IOT (through cellular or WIFI). When there is an error in one of the nx functions, for example in the "nx_tcp_server_socket_relisten" function, I get an error 0x23 which is NX_PORT_UNAVAILABLE. After that, it was impossible to continue exchanging through netcat on WSL, whatever the port. After a timeout of about 10 minutes, everything worked normally again.

[image]

[image]

In fact, the various communication buses are totally multi-threaded but they all seem to refer to the same instance of Netxduo since all the ports are blocked simultaneously and then unblocked at the same time. 

Do you have any idea which parameter I can change to avoid this problem? 

Best regards,
Antoine

TiejunMS, 2023-08-09:
0x23 should not be returned from `nx_tcp_server_socket_relisten`. Is the return status `0x35` instead? If so, it could be that the TCP server is not cleaned up before relisten. Try to call following APIs before it.
```
nx_tcp_socket_disconnect()
nx_tcp_server_socket_unaccept()
```

TiejunMS, 2023-08-10:
Do you mean 10 minutes delay by calling relisten? If not, could you describe the question again, thanks!

ZerAtaii, 2023-08-18:
In fact, sometimes i'm blocked for 29/30mins (i have measured it last week) and i don't know why, but one solution to this issue is to "deactivate/reactivate" my usb driver. The functions called are ux_slave_class_cdc_ecm_instance_activate and ux_slave_class_cdc_ecm_instance_deactivate, do you know how theses functions are triggered (HW or SF) ? Is there a way to reduce this strange 30mins delay to like 30secs / 1min?
[image]

TiejunMS, 2023-08-21:
I'm a little bit confused about the 'delay'. Where does it occur? In `nx_tcp_server_socket_relisten`, `nx_tcp_server_socket_accept` or somewhere else?

ZerAtaii, 2023-08-24:
[image]
After getting "Host disconnect" from TCP socket, i tried to nx_tcp_socket_disconnect()/nx_tcp_server_socket_unaccept()/nx_tcp_server_socket_relisten() as you said and that's working (no more error). But i don't have any interrupts from clients even if i try many times before 29 mins.

ZerAtaii, 2023-08-24:
And secondly, how can i trigger ux_slave_class_cdc_ecm_instance_activate and ux_slave_class_cdc_ecm_instance_deactivate from a software point of view?

TiejunMS, 2023-08-28:
@xiaocq2001 , can you help with USBX questions?

TiejunMS, 2023-11-24:
Closing as duplicate with https://github.com/azure-rtos/usbx/issues/114

closed 2023-11-24 by TiejunMS