eclipse-threadx/netxduo #189: Multiple issues on AP DHCPv6 mode

SwamidasSilabs, 2023-08-18:
**Bug**
- We observed two issues with DHCPv6 server functionality.

1) The first time when a client is connecting to the DHCPv6  server, we see DHCPv6 handshake messages (Solicit, Advertise, Request, Reply, Confirm, Reply). In these transactions, The **Confirm** packet coming from the client is having "Valid lifetime value 0" in the IA Address field.  In NetX code, this value (0) is loaded to a variable called "nx_dhcpv6_ia.nx_valid_lifetime" and Netx periodically checks the IP lease time with respect to this variable, So when the next periodic event comes to check the IP lease time we have one condition in the "_nx_dhcpv6_server_thread_entry" function and it is satisfying and clearing the connected client record from the table. Clearing this entry results in the same address being assigned to the next client that connects to the server.
- below is the piece of code that clears the entry by checking the condition.
- if (dhcpv6_client_ptr -> nx_dhcpv6_IP_lease_time_accrued >= dhcpv6_client_ptr -> nx_dhcpv6_ia.nx_valid_lifetime)
    {
        tx_mutex_put(&dhcpv6_server_ptr -> nx_dhcpv6_server_mutex);

        /* Clear the client record. */
        _nx_dhcpv6_clear_client_record(dhcpv6_server_ptr, dhcpv6_client_ptr);
    }
- nx_dhcpv6_IP_lease_time_accrued  is the time that periodically increases to reach the valid lifetime But where as the valid lifetime is set to 0.

2) The second issue comes when you disconnect the client and try to reconnect again (consider this is a 2nd cycle), At the time of reconnecting we see the Confirm, Reply messages. If you see in the 1st cycle the last message from the station is "Confirm" and in the 2nd cycle and 1st message from the station is "Confirm". So in the NetX code, We have a condition like if the last message is not equal to the current message type then update the client record, I am updating the piece of code below, So because of this the session timeout is happening in the "_nx_dhcpv6_server_thread_entry"  and clearing the client record entry.

if ((temp_client_rec.nx_dhcpv6_message_type != (*dhcpv6_client_ptr) -> nx_dhcpv6_message_type) &&
            (received_message_type != NX_DHCPV6_MESSAGE_TYPE_INFORM_REQUEST))

I am attaching the wireshark captures for your reference.
1) The Below picture shows the Confirm packet of the 1st cycle.
[image]

2) The Below picture shows the 2nd cycle.
[image]

TrumsZ, 2023-08-30:
Hello @SwamidasSilabs, sorry for the late response, thanks for bring-up the module and remind us of these issues.

1. For the first one, we would like to confirm that do you mean the DHCPv6 server should ignore and not update the value of the preferred-lifetime and valid-lifetime fields in IA_NA options in CONFIRM messages. If yes, we will fix it in the future release. If no, please provide more details for the issue you met.

2. For the second one, do you mean in the 2nd cycle, the replied xid (0xa44c3a) is not equivalent to the xid in CONFIRM message (0x631952). Could you also please provide more details such as what kind of code updating you have done to fix the issue as a workaround. Thank you very much.

SwamidasSilabs, 2023-09-12:
> Hello @SwamidasSilabs, sorry for the late response, thanks for bring-up the module and remind us of these issues.
>
> 1. For the first one, we would like to confirm that do you mean the DHCPv6 server [...]

TrumsZ, 2023-09-12:
Hi @SwamidasSilabs, did not see any comments in your replication.

SwamidasSilabs, 2023-09-12:
Hi @TrumsZ 
Sorry for the late.

1)  For the first issue, The server has to ignore the preferred lifetime and valid lifetime fields in the iana options, if they are set to 0.
I am attaching a snapshot of the RFC standards page and the link to it. https://www.ietf.org/rfc/rfc3315.txt

Below is a fix that I added for the 1st issue.

if(data) {
    dhcpv6_client_ptr -> nx_dhcpv6_ia.nx_valid_lifetime = data;
}
else {
    dhcpv6_client_ptr -> nx_dhcpv6_ia.nx_valid_lifetime = NX_DHCPV6_DEFAULT_VALID_TIME;
}

if(data) {
    dhcpv6_client_ptr -> nx_dhcpv6_ia.nx_preferred_lifetime = data;
}
else {
    dhcpv6_client_ptr -> nx_dhcpv6_ia.nx_preferred_lifetime= NX_DHCPV6_DEFAULT_PREFERRED_TIME;
}

2) For the 2nd issue, I am attaching my workaround, Please check that.

if (((temp_client_rec.nx_dhcpv6_message_type != (*dhcpv6_client_ptr) -> nx_dhcpv6_message_type) || (temp_client_rec.nx_dhcpv6_message_type == NX_DHCPV6_MESSAGE_TYPE_CONFIRM))&& 
             (received_message_type != NX_DHCPV6_MESSAGE_TYPE_INFORM_REQUEST))

Thanks,
Swamidas
[image]

TrumsZ, 2023-09-14:
Hi @SwamidasSilabs,

For the first issue, the RFC says the server MUST ignore any received preferred-lifetime and valid-lifetimes values in the message sent by a client to a server as the figure shown below. Therefore, it is probable that the correct behaviour for the server shall execute nothing for these 2 fields. How do you think about it?
[image]

bo-ms, 2023-11-01:
Hi @SwamidasSilabs The issues are fixed in latest version 6.3.0, could you try it?

bo-ms, 2023-11-01:
Closing, @SwamidasSilabs feel free to reopen or create new issue if you have further issue.

closed 2023-11-01 by bo-ms