eclipse-threadx/netxduo #190: How to troubleshooting PPP Link failure

dsilva-vd, 2023-08-31:
Hi,

I am running into a PPP Link failure using NetXDuo 6.2.0, STM32 hardware with Azure RTOS and ST Cellular middleware. I want to ask how to go start debugging the failure to see if it is PPP layer problem or not.

Here is my application:
1. NetXDuo release 6.2.0
2. STM32L4R9 chipset
3. Cellular modem with UART Interface to STM32 chip (host running NetX PPP stack)
4. ST x-cube-cellular-v7-0-0 cellular middleware

My application runs with successful PPP sessions using SIM from carrier A.
With SIM from different carrier B the PPP sessions starts but I encounter a PPP Link Down callback (set in nx_ppp_link_down_notify()) about 1 minute (I need to measure the exact time) into the session. From the carrier's website it shows the modem is still connected to the network. From the modem logs, the modem does not have an issue connecting to the network

This problem has occurred with NetXDuo release 6.0.7 and I recently upgraded to release 6.2.0 to retest. The behavior is still the same.

I have also tried the same modem and SIM (Carrier B, same physical location) in a different device (Linux SBC with different cellular middleware and PPP stack) and PPP sessions are successfully made. This tells me the issue is not related to signal strength at my location.  There is no PAP or CHAP authentication on either  Carrier A or Carrier B.

Where do I start to troubleshoot the problem? The link down callback has the NX_PPP variable but what information can I look for in that variable. 

Thanks

dsilva-vd, 2023-09-05:
For those who have the same question see nx_user.h

Enable the following defines:
```
#define NX_PPP_DEBUG_LOG_ENABLE
#define NX_PPP_DEBUG_LOG_PRINT_ENABLE
```

closed 2023-09-05 by dsilva-vd

hpat26, 2024-05-21:
THe Issue is stil being observed. 9 out of 10 times the DNS_RESOLVE_FAIL error is caused. Error # 163. SNTP Sync fails as well. Please find the attached file containing ppp logs
[DNS RESOLVE FAILED.txt](https://github.com/eclipse-threadx/netxduo/files/15395260/DNS.RESOLVE.FAILED.txt)
[DNS RESOLVE SUCCESS.txt](https://github.com/eclipse-threadx/netxduo/files/15395261/DNS.RESOLVE.SUCCESS.txt)
[SNTP SYNC FAILED.txt](https://github.com/eclipse-threadx/netxduo/files/15395262/SNTP.SYNC.FAILED.txt)