eclipse-threadx/netxduo #191: Decoding PPP LInk failure from PPP Debug Logs

dsilva-vd, 2023-09-05:
Hi,

I am running into a PPP Link failure using NetXDuo 6.2.0, STM32 hardware with Azure RTOS and ST Cellular middleware. I have captured a PPP Debug log but I would like help to understand the problem

Here is my application:

NetXDuo release 6.2.0
STM32L4R9 chipset
Cellular modem with UART Interface to STM32 chip (host running NetX PPP stack)
ST x-cube-cellular-v7-0-0 cellular middleware
My application runs with successful PPP sessions using SIM from carrier A.
With SIM from different carrier B the PPP sessions starts but I encounter a PPP Link Down callback (set in nx_ppp_link_down_notify()) about 1 minute (I need to measure the exact time) into the session. From the carrier's website it shows the modem is still connected to the network. From the modem logs, the modem does not have an issue connecting to the network

This problem has occurred with NetXDuo release 6.0.7 and I recently upgraded to release 6.2.0 to retest. The behavior is still the same.

I have also tried the same modem and SIM (Carrier B, same physical location) in a different device (Linux SBC with different cellular middleware and PPP stack) and PPP sessions are successfully made. This tells me the issue is not related to signal strength at my location. There is no PAP or CHAP authentication on either Carrier A or Carrier B.

Thanks

PPP Debug Log file attached is from carrier B. I can share a PPP Debug log from carrier A if it is useful

[Multi-carrier SIM SWV with PPP Debug enabled - PPP Debug only.txt](https://github.com/azure-rtos/netxduo/files/12526923/Multi-carrier.SIM.SWV.with.PPP.Debug.enabled.-.PPP.Debug.only.txt)

dsilva-vd, 2023-09-05:
Carrier B is an MVNO for Carrier A's network so I do not believe the cellular signal conditions are an issue.

bo-ms, 2023-09-25:
Hi @dsilva-vd Sorry for the late response. From your log, PPP server sent [Terminate-Request](https://datatracker.ietf.org/doc/html/rfc1661#section-5.5) (c0 21 **05** 04 00 04) to device to close the connection, so NetX Duo PPP invoked the link down callback. 

```
Time: 13684, PPP Name: PPP Instance, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated, Send Packet Length: 62, Packet: 00 21 45 00 00 3c 00 06 40 00 80 11 24 f0 ac 10 19 9b 08 08 08 08 e0 f0 00 35 00 28 78 c3 38 f8 01 00 00 01 00 00 00 00 00 00 01 31 04 70 6f 6f 6c 03 
---->>>>> PPP LInk down callback occurred 
Time: 14922, PPP Name: PPP Instance, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated, Received Packet Length: 6, Packet: c0 21 05 04 00 04 
Time: 14922, PPP Name: PPP Instance, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated, Send Packet Length: 6, Packet: c0 21 06 04 00 04 
```

bo-ms, 2023-11-01:
Closing due to no response, @dsilva-vd feel free to create new issue or reopen it if you have further issues.

closed 2023-11-01 by bo-ms

hpat26, 2024-05-21:
The Issue is stil being observed. 9 out of 10 times the DNS_RESOLVE_FAIL error is caused. Error # 163. SNTP Sync fails as well. Please find the attached file containing ppp logs
[DNS RESOLVE FAILED.txt](https://github.com/eclipse-threadx/netxduo/files/15395291/DNS.RESOLVE.FAILED.txt)
[DNS RESOLVE SUCCESS.txt](https://github.com/eclipse-threadx/netxduo/files/15395292/DNS.RESOLVE.SUCCESS.txt)
[SNTP SYNC FAILED.txt](https://github.com/eclipse-threadx/netxduo/files/15395293/SNTP.SYNC.FAILED.txt)