eclipse-threadx/netxduo #192: UDP notification and queue depth feature.

mika-saf, 2023-09-07:
Hello,

We seems to have founded an issue on UDP notification.
In our application, we set a maximum queue depth for UDP socket, and we activate a notification when a frame is received.
In this particular case, we have seen that when the queue is full, we receive also a notification for a UDP frame, but the frame is dropped. It should not receive a notification in this case.

We can see in file nx_udp_packet_receive.c that when the max size is reached the packet is dropped (line 492/493) but the notification is called (see line 545).

version: netxduo 6.1.9
target: xilinx ultrascale+

**To Reproduce**
Steps to reproduce the behavior:
1. declare a UDP socket, set a max depth and activate notification.
1. receive a lot of UDP packet, check notification, if max depth is reached, a notification will occurs but no socket linked, 

**Expected behavior**
No notification must be occurs when packet is dropped.

Here attached, a suggestion for fixing.

[26.patch](https://github.com/azure-rtos/netxduo/files/12551137/26.patch)

mika.

wenhui-xie, 2023-11-13:
Hi @mika-saf, if the max queue size is reached, the oldest packet will be dropped and the new packet will be added to the queue. It doesn't make sense if no notification in this situation.

mika-saf, 2023-11-14:
Hello,
In the notification we don't have the status (packet lost or not), so we can try to unstack packet but it doesn't exist.

wenhui-xie, 2023-11-16:
Hi @mika-saf, when the max queue size is reached, we need to drop the oldest packet. This is the established logic. For the notification function, there are some situations in which users receive the notification but can not get a packet(e.g. packet with error checksum). So it's reasonable to notify users of the new packet even if the old one may be dropped.

wenhui-xie, 2023-11-22:
The notification function is an API, so we will not change the parameters of this function.

And you can get the number of the dropped packets using API nx_udp_socket_info_get() which will help you to count the total received packets.

bo-ms, 2023-12-04:
Closing due to no further response, feel free to create new issue if you encounter any issues.