eclipse-threadx/netxduo #194: Getting hardfault in _nx_packet_release function

kaushaleinfo, 2023-09-19:
Hi all,
I am using stm32f205 mcu and using threadx, netx 6.2 version,
Random time I am getting hardfault from **_nx_packet_release** so I did depth debugging and found where exact it is generating hardFault,
So it is generating while   **/* Update the links of the adjacent threads.  */** in _nx_packet_release function, 
 
/* Determine if there are any threads suspended on the block pool.  */
        thread_ptr =  pool_ptr -> nx_packet_pool_suspension_list;
        if (thread_ptr)
        {
          /* Remove the suspended thread from the list.  */

            /* See if this is the only suspended thread on the list.  */
            if (thread_ptr == thread_ptr -> tx_thread_suspended_next)
            {

                /* Yes, the only suspended thread.  */

                /* Update the head pointer.  */
                pool_ptr -> nx_packet_pool_suspension_list =  NX_NULL;
            }
            else
            {

                /* At least one more thread is on the same expiration list.  */ 

                /* Update the list head pointer.  */
                pool_ptr -> nx_packet_pool_suspension_list =  thread_ptr -> tx_thread_suspended_next;

----------->>>> from below line code it is generating a hardFault, <<<<------------------
 
                **/* Update the links of the adjacent threads.  */
                (thread_ptr -> tx_thread_suspended_next) -> tx_thread_suspended_previous =
                    thread_ptr -> tx_thread_suspended_previous;**
-----------------------------------------------------------------------------------------------------

Mostly it is generating after discover DHCP packet, Could you guys tell me why this is happening or where should I look?

TiejunMS, 2023-11-24:
It sounds like memory corruption or illegal usage of packet. Could you try to disable all logics from application and run DHCP only? See if the problem is still there.

TiejunMS, 2023-12-28:
Closing as no response.

closed 2023-12-28 by TiejunMS

mo-krauti, 2025-05-13:
Im am trying to get the Ethernet to work on an NUCLEO-H753ZI Board using NetXDuo and running into exactly this hardfault. 
Relevant linker script section:
```
  .eth_sec (NOLOAD) :
  {
    . = ABSOLUTE(0x30040000);
    *(.RxDescripSection)

    . = ABSOLUTE(0x30040060);
    *(.TxDescripSection)

    . = ABSOLUTE(0x30040200);
    *(.NetXPoolSection)

  } >RAM_D2 AT> FLASH
```
MPU configuration:
```c
void MPU_Config(void)
{
  MPU_Region_InitTypeDef MPU_InitStruct = {0};

  /* Disables the MPU */
  HAL_MPU_Disable();

  /** Initializes and configures the Region and the memory to be protected
  */
  MPU_InitStruct.Enable = MPU_REGION_ENABLE;
  MPU_InitStruct.Number = MPU_REGION_NUMBER0;
  MPU_InitStruct.BaseAddress = 0x0;
  MPU_InitStruct.Size = MPU_REGION_SIZE_4GB;
  MPU_InitStruct.SubRegionDisable = 0x87;
  MPU_InitStruct.TypeExtField = MPU_TEX_LEVEL0;
  MPU_InitStruct.AccessPermission = MPU_REGION_NO_ACCESS;
  MPU_InitStruct.DisableExec = MPU_INSTRUCTION_ACCESS_DISABLE;
  MPU_InitStruct.IsShareable = MPU_ACCESS_SHAREABLE;
  MPU_InitStruct.IsCacheable = MPU_ACCESS_NOT_CACHEABLE;
  MPU_InitStruct.IsBufferable = MPU_ACCESS_NOT_BUFFERABLE;

  HAL_MPU_ConfigRegion(&MPU_InitStruct);

  /** Initializes and configures the Region and the memory to be protected
  */
  MPU_InitStruct.Number = MPU_REGION_NUMBER1;
  MPU_InitStruct.BaseAddress = 0x30040000;
  MPU_InitStruct.Size = MPU_REGION_SIZE_128KB;
  MPU_InitStruct.SubRegionDisable = 0x0;
  MPU_InitStruct.AccessPermission = MPU_REGION_FULL_ACCESS;
  MPU_InitStruct.IsBufferable = MPU_ACCESS_BUFFERABLE;

  HAL_MPU_ConfigRegion(&MPU_InitStruct);
  /* Enables the MPU */
  HAL_MPU_Enable(MPU_PRIVILEGED_DEFAULT);

}
```
NetXDuo initializes just fine without any error returned. The NetXDuo initialization code I wrote was working on an stm32 f767zi before, so I do not think the issue is related to that. 
I can even ping the board for about 3-4 times and than it always stops working. 
I also cannot reliable reproduce the hardfault right now, sometimes it just stops working without the hardfault.