eclipse-threadx/netxduo #198: X.509 Chain Certificates

w-samuelgauthier, 2023-10-19:
Hi all,

I'm trying to provision my STM32BL4S5IIOT01A board to the IoT Hub via Group Provisioning using X.509 certificates. I am using the [getting-started](https://github.com/azure-rtos/getting-started) sample and an Ubuntu VM. However this doesn't work.

The problem is that netxduo expects PKCS#1 formatted X.509 DER certificates and this format does not support chain certificates.

How can I use netxduo and chain certificates?

Just for info I successfully run the X.509 individual enrollment [explained here](https://github.com/azure-rtos/getting-started/blob/master/docs/using-self-signed-x509-certificates.md) and the X.509 group enrollment [explained here](https://learn.microsoft.com/en-us/azure/iot-dps/tutorial-custom-hsm-enrollment-group-x509?tabs=linux&pivots=programming-language-ansi-c) with the azure-iot-sdk-c

TiejunMS, 2023-10-20:
NetX Duo does not support sending cert chain for device auth. For group enrollment, you can enroll the intermedium ca cert. Then from device side, you will only need to send leaf cert for auth.

w-samuelgauthier, 2023-10-20:
Thank you very much for clarifying!

I tried the following but without success:
1. create a cert chain `root>intermediate>device` like explained [here](https://learn.microsoft.com/en-us/azure/iot-dps/tutorial-custom-hsm-enrollment-group-x509?tabs=linux&pivots=programming-language-ansi-c)
2. Transform the device private key and cert into DER format and add them to the device
3. Create an enrollment group in DPS with the intermediate cert

I also tried to add the `root` and `intermediate` certs to the DPS (under *Certificates*) but it still doesn't work (it's stuck the Pending state).

Is there a way to log the network (mqtt) traffic with netxduo? How can I see what's happening?

TiejunMS, 2023-10-26:
There is not much useful information from device side if the device is stuck at the pending state. Please reach out to the support team of DPS service. They can help you get the root cause of why the device can not finish the registration.

But one more simple check I can think of, check whether you have verified the ownership of the CA certs. https://learn.microsoft.com/en-us/azure/iot-dps/tutorial-custom-hsm-enrollment-group-x509?tabs=linux&pivots=programming-language-ansi-c#verify-ownership-of-the-root-certificate

w-samuelgauthier, 2023-10-31:
Thank you for your answer. Yes, I have also verified the ownership of the certificates, but it is still the same.

TiejunMS, 2023-11-01:
Could you define `NX_AZURE_IOT_LOG_LEVEL` to `3`, recompile NetX Duo library and run the application again? Let us know the output from DPS client.

TiejunMS, 2023-11-24:
Closing as no response

closed 2023-11-24 by TiejunMS