eclipse-threadx/netxduo #201: Notify MQTT_CONTROL_PACKET_TYPE_PUBACK event to user

ZerAtaii, 2023-10-26:
Hi ! 
I use NetxDuo MQTT Client to exchange data with the Azure cloud. Currently, when sending an MQTT message, the application returns an OK to the user via the "nx_azure_iot_publish_mqtt_packet" function. Would it be possible to have feedback when the "MQTT_CONTROL_PACKET_TYPE_PUBACK" event is received? Does a callback method already exist?

Best regards,
Antoine

bo-ms, 2023-10-27:
Hi @ZerAtaii Thanks for reaching out to us. Currently, Azure RTOS does not have callback to track PUBACK. We will check this feature internally and keep you updated.

bo-ms, 2023-11-24:
Hi @ZerAtaii A new callback will be added for tracking telemetry ack, I will let you know when we publish the code on GitHub.

bo-ms, 2023-12-04:
Hi @ZerAtaii The new feature of tracking telemetry ack is added in #214

closed 2023-12-04 by bo-ms

ZerAtaii, 2023-12-04:
Hi @bo-ms, 
Thank you for the job :)