eclipse-threadx/netxduo #202: [ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541

Clyasenth, 2023-10-31:
Hi I am using the b_u585i_iot02a board for the Azure-IOT-Central application. I applied all of the instructions to execute the NetXDuo/Nx_Azure_IoT application but I am stuck at this point.

ATCModem:Tempo started (150000 ms)...

ATCModem:Tempo started (150000 ms)...

ATCore ERROR:AT_sendcmd error

ATCore ERROR:AT_sendcmd error: parse from rsp

LOG ERROR #3: channel=1 / errorId=11 / gravity=2

LOG ERROR #4: channel=1 / errorId=14 / gravity=2

[ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541

Disconnected from IoTHub!: error code = 0x00010005

Reconnecting iothub, after backoff

ATCModem:Tempo started (150000 ms)...

ATCModem:Tempo started (150000 ms)...

Could someone help me to fix this?

bo-ms, 2023-11-01:
Hi @Clyasenth Thank you for reaching out to us. Looks the issue is for cellular module (**ATCore ERROR**), have you connected the cellular module to u5? You may also ask for help from ST: https://community.st.com/t5/stm32-mcus-embedded-software/bd-p/mcu-embedded-software-forum

Clyasenth, 2023-11-01:
Hi @bo-ms thanks for the quick reply. Yeah, I connected the cellular module to u5 which uses bg96. This log appears after the cellular module initialization.

Init sensors on board:

        Init magnetometer IIS2MDC sensor

        Init gyroscope and accelerometer ISM330DHCX sensor

        Init temperature and humidity HTS221 sensor

        Init temperature and pressure LPS22HH sensor

        Enabled magnetometer IIS2MDC sensor

        Enabled accelerometer ISM330DHCX sensor

        Enabled gyroscope ISM330DHCX sensor

        Enabled temperature HTS221 sensor

        Enabled humidity HTS221 sensor

        Enabled pressure LPS22HH sensor

Init Full Scale 4G for accelerometer ISM330DLC

Init Full Scale 250DPS for gyroscope ISM330DLC

[INFO] Azure IoT Security Module has been enabled, status=0

IoTHub Host Name: *****;

Device ID: *****.

ATCModem:Tempo started (150000 ms)...

ATCModem:Tempo started (150000 ms)...

ATCore ERROR:AT_sendcmd error

ATCore ERROR:AT_sendcmd error: parse from rsp

LOG ERROR #1: channel=1 / errorId=11 / gravity=2

LOG ERROR #2: channel=1 / errorId=14 / gravity=2

[ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541

Disconnected from IoTHub!: error code = 0x00010005

Reconnecting iothub, after backoff

ATCModem:Tempo started (150000 ms)...

bo-ms, 2023-11-02:
Hi @Clyasenth I am not quite sure if the AT_sendcmd error impacts it. For IoT Central application, I guess you use certificate for connection, correct? could you check if the certificate is set correctly in your project? Where do you get this project? if ST provides this project, you may contact them. Thanks.

bo-ms, 2023-11-24:
Closing due to no further response, feel free to reopen the issue

closed 2023-11-24 by bo-ms