eclipse-threadx/netxduo #204: HTTP 3xx Location redirects are discarded

vnp85, 2023-11-03:
Although status codes like 0x30023 (http code 301 moved permanently) do appear as a result code, the `response header callback` is not called by the function if not `http 2xx`, thus the Location header's content doesn't reach the client code.

https://github.com/azure-rtos/netxduo/blob/531239b44a13845130ff87e9632d9de61d072177/addons/web/nx_web_http_client.c#L1637C1-L1638C1

wenhui-xie, 2023-11-09:
Hi @vnp85, thanks for your report. What is the scenario in which you encountered this issue? Is this issue in your test project or your official product?

vnp85, 2023-11-09:
Step1: set up a webserver. Say Apache. We then have something like `public_html/folder-of-interest/index.php` which is accessible, through a web browser, at the address `http://example.com/folder-of-interest` -- however, browsers (and curl and wget etc) follow 3xx redirects, and so we end up at `http://example.com/folder-of-interest/` - notice the tailing slash has been added
Step2: set up the IoT http web client, to access `http://example.com/folder-of-interest` 
Step3: Observe in Wireshark, it indicates the traffic happens as expected, with `301 Moved permanently` present in the server response, Location, all there. 

Expected: the NX code to expose the Location header (among the other response headers)
Observed: the NX code does return an internal status code, for example 0x30023 NX_WEB_HTTP_STATUS_CODE_MOVED_PERMANENTLY, but because of the highlighted line of code, the header-digesting function `_nx_web_http_client_process_header_fields` which would call the client callback, is not called.

It's a project I am tinkering with.

wenhui-xie, 2023-11-13:
@vnp85 Sorry that we don't support processing response headers for 3xx code.

closed 2023-11-22 by wenhui-xie