eclipse-threadx/netxduo #206: Issue in DHCPv6 Client

swamidas1998, 2023-11-06:
**Bug**
We observed one issue when running the dhcpv6 client.

1) In the DHCPv6 client code, We saw an issue in which the SOLICIT request message is been continuously sent infinite times if the client doesn't get an ADVERTISE reply for a SOLICIT request message. Due to this, the DHCPv6 continues the process forever and never returns.

2) My observation here is, **nx_dhcpv6_max_retransmission_count** This variable is not updating correctly, During SOLICIT this variable is set to 0 and so as the 0 is ignored in the **_nx_dhcpv6_update_retransmit_info** and not considering the count of SOLICIT messages stored in the **nx_dhcpv6_retransmission_count**

Thanks,
Swamidas

wenhui-xie, 2023-11-06:
Hi @swamidas1998, nx_dhcpv6_max_retransmission_count is set to NX_DHCPV6_MAX_SOL_RETRANSMISSION_COUNT. The default value of NX_DHCPV6_MAX_SOL_RETRANSMISSION_COUNT is 0, which means no limit to the retransmission count. You can redefine the NX_DHCPV6_MAX_SOL_RETRANSMISSION_COUNT in nx_user.h or project options to enable the retransmission limit.

swamidas1998, 2023-11-06:
Hi, @wenhui-xie I don't see the macro **NX_DHCPV6_MAX_SOL_RETRANSMISSION_COUNT** is assigned to the **nx_dhcpv6_max_retransmission_count** variable anywhere.

wenhui-xie, 2023-11-07:
Hi @swamidas1998, which version are you using? For the latest release, the assignment of **nx_dhcpv6_max_retransmission_count** is in function _nx_dhcpv6_request_solicit [line 7943](https://github.com/azure-rtos/netxduo/blob/531239b44a13845130ff87e9632d9de61d072177/addons/dhcp/nxd_dhcpv6_client.c#L7943C5-L7943C95).

closed 2023-11-22 by wenhui-xie