eclipse-threadx/netxduo #207: Not able to compile add_on->Azure_iot

kalijana, 2023-11-08:
Azure RTOS with netx compliles , when trying to add azure_iot in add-ons not able to compile 

Please also mention any information which could help others to understand
the problem you're facing:
```
--->Not able to compile the following files 
nx_azure_iot.c
nx_azure_iot.h
nx_azure_iot_adu_agent.c
nx_azure_iot_adu_agent.h
nx_azure_iot_adu_root_key.c
nx_azure_iot_hub_client.c
nx_azure_iot_hub_client.h
nx_azure_iot_hub_client_properties.c
nx_azure_iot_hub_client_properties.h
nx_azure_iot_json_reader.c
nx_azure_iot_json_reader.h
nx_azure_iot_json_writer.c
nx_azure_iot_json_writer.h
nx_azure_iot_provisioning_client.c
nx_azure_iot_provisioning_client.h

```
- What target device are you using?
`-> STM32MCU -> L4 `
- Which version of Azure RTOS?
- `latest`
- What toolchain and environment?
```
-> STMicroelectronics	STM32CubeMX	6.9.2.202309101412	
- >Eclipse CDT	CDT GCC Cross Compiler Support	11.1.1.202303201356	org.eclipse.cdt.build.crossgcc
```
- What have you tried to diagnose or workaround this issue?
`how to compile these files `

bo-ms, 2023-11-09:
Hi @kalijana Could you paste the compilation errors? thanks.

bo-ms, 2023-11-24:
Closing due to no further response, feel free to reopen the issue.

closed 2023-11-24 by bo-ms