eclipse-threadx/netxduo #221: SNMP Agent is failing while processing the Version-3 SET request with GCC 12.2(and above).

bhavik-patel-vf, 2023-12-19:
**Describe the bug**
When the SNMP Version-3 SET request is received from the SNMP manager, it executes the **_nx_snmp_version_3_process** API to process the SNMP v3 request, inside that **_nx_snmp_utility_object_data_get** API got called and it's API return value got corrupted.
With **GCC  v10.3.** version-3 SET request was processed successfully but with **GCC v12.2** it is failing.

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using?
> **Renesas EK-RA6M5**(R7FA6M5BH3CFC)
- Which version of Azure RTOS?
> **v6.2.1**
- What toolchain and environment?
> **GCC v12.2**, E2 studio
- What have you tried to diagnose or workaround this issue?
> Debugged the **_nx_snmp_utility_object_data_get** API while execution, as per the image given below in the disassembly window it can be seen that in between **ZERO** value returned.
[image]
[image]

**Additional context**
> **With **GCC  v10.3.** version-3 SET request was processed successfully but with **GCC v12.2** it is failing.**

bhavik-patel-vf, 2024-01-31:
Hi @wenhui-xie ,
any update on this ticket?

CC: @bo-ms

bhavik-patel-vf, 2024-03-29:
Hi @wenhui-xie  ,
Please provide any update on this ticket.

CC: @bo-ms

bhavik-patel-vf, 2024-06-12:
Reminder,

Hi @wenhui-xie ,
Kindly please provide any update on this ticket.

CC: @bo-ms

prasannavk-eic, 2025-02-03:
Hi @wenhui-xie 
Kindly please provide any update on this ticket.

CC: @bo-ms

sravani-25-03-2003, 2025-04-21:
Any body , please provide the configurations of snmp (azzure rtos) step by step .