eclipse-threadx/netxduo #234: Tcp socket receive problem for large messages

chrishoen, 2024-01-24:
I'm having problems with a thread that receives large tcp messages (100 KB). 
The thread uses a tcp socket receive notification callback that reads packets
into a buffer. Because the documentation on how to do this isn't very good,
I was wondering if you could verify that the following looks okay:

void TcpServerThread::onRxNotify1(NX_TCP_SOCKET* aSocket)
{
   UINT tRet = 0;
   NX_PACKET* tPacket = 0;
   ULONG tRxSize = 0;

   // Loop to read tcp packet chains into the receive socket buffer until
   // the receive queue is empty.
   while (1)
   {
      // Nonblocking receive a packet chain.
      tRet = nx_tcp_socket_receive(&mSocket, &tPacket, NX_NO_WAIT);

      // Check.
      if (tRet == NX_NO_PACKET)
      {
         // The receive queue is empty.
         trc_write("recv1 NX_NO_PACKET");
         break;
      }

      // Check.
      if (tRet == NX_NOT_CONNECTED)
      {
         trc_write("recv1 NX_NOT_CONNECTED", tRet);
         return;
      }

      // Check.
      if (tRet != NX_SUCCESS)
      {
         trc_write("recv1 FAIL1", tRet);
         return;
      }

      // Extract bytes from the packet chain into the receive socket buffer with
      // the total size as an offset.
      tRet = nx_packet_data_retrieve(tPacket, &mRxSocketBuffer[mRxTotalSize], &tRxSize);

      // Advance the total size offset.
      mRxTotalSize += tRxSize;
      trc_write("recv1 %d", tRxSize);

      // Release the packets.
      nx_packet_release(tPacket);
   }

   // Done extracting into the buffer.
   trc_write("recv1 done %d", mRxTotalSize);
}

Thanks
Chris

TiejunMS, 2024-01-25:
Your code looks good. What problem do you encounter?

Hnz2, 2024-01-25:
Hi,

I think code is not 100% correct, because it does not check buffer _mRxSocketBuffer[]_ for overflow. I think it should be called _nx_packet_length_get()_ first to allow overflow check.

Jan

chrishoen, 2024-01-25:
Thanks guys.

1. I just wanted to see if I was doing things correctly with the receive call, the documentation for that is not very helpful.  
2. I'm not worried about overflow here, that's just an example snippet.
3. As far as what problems am I encountering, that's a long story. Too hard to describe here.
4. Thanks for looking at it.

Chris

closed 2024-01-25 by chrishoen