eclipse-threadx/netxduo #255: BSD addon fails to compile

sj-raima, 2024-03-05:
v6.4.1_rel of netxduo, filex, and threadx
Hosted build for Linux - Ubuntu 20.04.6 LTS (GNU/Linux 5.15.0-97-generic x86_64)

Nested filex inside this project as follows:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index c490b3c..f9f1b6b 100755
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -5,6 +5,8 @@ project(netxduo
     LANGUAGES C ASM
 )    
 
+add_subdirectory (filex)
+
 option(NXD_ENABLE_FILE_SERVERS "Includes a dependency on FileX to support 'server' protocol handlers" ON)

```
and similarly for threadx inside filex.

Running cmake as follows:

```
$ mkdir build && cd build
$ cmake -DCMAKE_BUILD_TYPE=Debug -DNXD_ENABLE_BSD=yes -DTHREADX_ARCH=linux -DTHREADX_TOOLCHAIN=gnu ..

```
Make fails with the following errors:

```
[ 31%] Building C object CMakeFiles/netxduo.dir/addons/BSD/nxd_bsd.c.o
In file included from /home/sj/netxduo/addons/BSD/nxd_bsd.c:28:
/home/sj/netxduo/addons/BSD/nxd_bsd.h:628:21: error: typedef redefinition with different types ('LONG' (aka 'int') vs '__suseconds_t' (aka 'long'))
typedef LONG        nx_bsd_suseconds_t;
                    ^
/home/sj/netxduo/addons/BSD/nxd_bsd.h:208:33: note: expanded from macro 'nx_bsd_suseconds_t'
#define nx_bsd_suseconds_t      suseconds_t
                                ^
/usr/include/x86_64-linux-gnu/sys/types.h:138:23: note: previous definition is here
typedef __suseconds_t suseconds_t;
                      ^
In file included from /home/sj/netxduo/addons/BSD/nxd_bsd.c:28:
/home/sj/netxduo/addons/BSD/nxd_bsd.h:631:8: error: redefinition of 'timeval'
struct nx_bsd_timeval
       ^
/home/sj/netxduo/addons/BSD/nxd_bsd.h:210:33: note: expanded from macro 'nx_bsd_timeval'
#define nx_bsd_timeval          timeval
                                ^
/usr/include/x86_64-linux-gnu/bits/types/struct_timeval.h:8:8: note: previous definition is here
struct timeval
       ^
In file included from /home/sj/netxduo/addons/BSD/nxd_bsd.c:28:
/home/sj/netxduo/addons/BSD/nxd_bsd.h:703:3: error: typedef redefinition with different types ('struct FD_SET_STRUCT' vs 'struct fd_set')
} nx_bsd_fd_set;
  ^
/home/sj/netxduo/addons/BSD/nxd_bsd.h:226:33: note: expanded from macro 'nx_bsd_fd_set'
#define nx_bsd_fd_set           fd_set
                                ^
/usr/include/x86_64-linux-gnu/sys/select.h:70:5: note: previous definition is here
  } fd_set;
    ^
```

Adding this define:

```
diff --git a/addons/BSD/nxd_bsd.h b/addons/BSD/nxd_bsd.h
index 3810a81..e2b0460 100644
--- a/addons/BSD/nxd_bsd.h
+++ b/addons/BSD/nxd_bsd.h
@@ -135,9 +135,7 @@ extern   "C" {
 
 
 /* Defined, APIs are natively named with nx_bsd_ prefix to avoid conflicting/overriding OS BSD APIs.  */
-/*
 #define NX_BSD_ENABLE_NATIVE_API
-*/

```

gives instead these errors:

```
/home/sj/netxduo/addons/BSD/nxd_bsd.c:10121:29: error: no member named 'bsd_errno' in 'struct TX_THREAD_STRUCT'
      current_thread_ptr -> bsd_errno = tx_errno;
      ~~~~~~~~~~~~~~~~~~    ^
/home/sj/netxduo/addons/BSD/nxd_bsd.c:10180:33: error: no member named 'bsd_errno' in 'struct TX_THREAD_STRUCT'
    val = current_thread_ptr -> bsd_errno;
          ~~~~~~~~~~~~~~~~~~    ^

```

rahmanih, 2024-03-05:
hi @sj-raima,
you'll need to add a define in the 'tx_user.h' file as below:
https://github.com/eclipse-threadx/netxduo/blob/18704ffa7383071ce1b335730fbde131f27be045/test/cmake/libs/tx_user.h#L53

closed 2024-03-08 by rahmanih

sj-raima, 2024-03-09:
@rahmanih The solution you proposed was enough to get me on track but it is really not appropriate to define this in a test.  There should instead be a define to enable this (NX_BSD or similar)  and I would suggest code in threadx/common/inc(_smp)/tx_api.h to add the declaration:

```
#ifdef NX_BSD
    int bsd_errno;
#endif
```

rubenax97, 2024-09-13:
Also, I would like to add that if you are using `errno.h` header file in the implementation, you will find incompatibility error due a definition of error code in both files, `nxd_bsd.h` and `errno.h`.
Keep an eye on that 👀